micropython/micropython #8326: socket.connect() blocks on esp32

AmirHmZz, 2022-02-19:
I'm trying to connect to a TCP socket server. My code works properly when the destination server is listening, but when my server goes down and not running, `socket.connect()` blocks about 10 seconds even with `socket.settimeout(2)`.

```Python
import socket
async def test() :
    ai = socket.getaddrinfo(host, port, 0, socket.SOCK_STREAM)[0]
    s = socket.socket(ai[0], ai[1], ai[2])
    s.settimeout(2)
    s.connect(ai[-1]) # This blocks for 10 seconds
    ...
```

Is there any other way to set timeout for `socket.connect()` ? I've also tried `uasyncio.wait_for()` to call `test()` with timeout but that is not working too.

I should mention that I'm using [this release](https://micropython.org/resources/firmware/esp32-20220117-v1.18.bin) which is `micropython 1.18` based on `IDF 4.2` on a `Wemos D1 R32` board which uses `ESP32-WROOM-32`.

AmirHmZz, 2022-02-21:
Maybe related to this [issue](https://github.com/espressif/esp-idf/issues/8296) in `esp-idf` project, Any opinion ?

IhorNehrutsa, 2022-03-24:
Forum link:
https://forum.micropython.org/viewtopic.php?f=18&t=12019

nedoskiv, 2022-07-27:
I can confirm that problem, on MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32

`>>> import usocket as socket
>>> sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
>>> sock.settimeout(2)
>>> sock.connect(("192.168.4.2",80))		// take more than 15 seconds
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 113] ECONNABORTED`

``

sosi-deadeye, 2022-07-27:
Confirmed.
```
/home/andre> repl
Entering REPL. Use Control-X to exit.
>
MicroPython v1.19.1-216-g45ab801-dirty on 2022-07-27; ESP32 module with ESP32
Type "help()" for more information.
>>>
>>>
>>> import usocket as socket
>>> sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
>>> sock.settimeout(2)
>>> sock.connect(("5.9.16.42",80))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 113] ECONNABORTED
>>>
```

The host 5.9.16.42 does not exist/respond, which I used for the test.
The controller is online behind a NAT.

`ECONNABORTED` is raised after ~7 seconds.

Simanas, 2023-04-13:
Up!

abegines, 2023-04-17:
Same "annoying" problem here with "MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32"

With "Chip is ESP32-D0WD-V3 (revision v3.0)" and same with "Chip is ESP32-D0WDQ6 (revision v1.0)"

Do you think this is a problem in micropython porting to ESP32 or an ESP_IF firmware bug?

FeroxTL, 2023-07-10:
Could be solved with polling:

```
timeout = 5  # seconds
server, port = 'localhost', 8000

sock = socket.socket()
addr = socket.getaddrinfo(server, port)[0][-1]

if sys.platform == 'linux':
    self.sock.settimeout(timeout)
    self.sock.connect(addr)
else:
    self.sock.setblocking(False)
    poller = select.poll()
    poller.register(self.sock, select.POLLIN | select.POLLOUT)

    try:
        self.sock.connect(addr)
    except OSError as e:
        # 119 == still connecting
        if e.errno != 119:
            raise e

    res = poller.poll(timeout * 1000)
    if not res:
        sock.close()
        raise OSError('Timeout')

    # Socket connected
    poller.unregister(sock)
    del poller
    del res
    gc.collect()
    sock.settimeout(timeout)

# sock is connected - use as usual -- sock.write(msg), sock.read(4) etc..
```

jamedeus, 2023-08-18:
Issue still present in micropython 1.20 on ESP32:

```
MicroPython v1.20.0 on 2023-04-26; ESP32 module with ESP32
Type "help()" for more information.
>>>
>>> import socket
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
True
>>> wlan.connect('ssid', 'password')
>>>
>>> s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
>>> s.settimeout(1)
>>> s.connect(('1.2.3.4', 80))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 113] ECONNABORTED
>>>
```

As others have noted there is an 18-19 second delay before ECONNABORTED. This happens with or without settimeout.

zcattacz, 2023-11-06:
Reproducable on 1.21.0, but tehe @FeroxTL's `select.poll` workaround works.

dpgeorge, 2023-11-06:
The problem seems to be that lwip does not support a timeout when setting up a connection.  See this comment: https://github.com/espressif/esp-idf/issues/8296#issuecomment-1094688334

The solution is as suggested there, to make the socket non-blocking and poll/select on the socket until the connection completes, or the timeout is reached.  That's how `asyncio.open_connection()` works.

I'm not sure if this should be fixed at the C level or not.  It may be that it is already correct Python behaviour, because it's following the system behaviour.

zcattacz, 2023-11-07:
To an unfamiliar user like me, this `connect` behavior is unexpected. `connect` in CPython3 on Linux honors `settimeout` set before `connect`.

Host connectivity is usually fairly good in coding phase, the 10s+ `connect` lockup could easily bite inexperienced users after early dev phase, and make things confusing when mixed with other timing issues.

I was testing for blocking socket, so I didn't even think about non-blocking socket which is polled and switch back with `settimeout in the workaround.

I think at least documenting this behavior mismatch with cpython could be worthwhile.

dpgeorge, 2023-11-08:
We should fix this.

zcattacz, 2023-11-08:
After some testing, I noticed a minor mismatch with the above workaround. The ECONNRESET is not raised on `connect`, but on first `sock.write()`, when connect dest is a non-open port of a live host (in wireshark you can see the RST pkt has already been send to mpy on connect before so.write).
It seem that checking `sock.fileno()>0` on micropython can avoid this though the exact failure cause is not clear.

On CPython, `connect` raise `ConnectionRefusedError` in this case, before any write attempt. this saves some redundant `try...catch...` on following `sock.write`s.

dpgeorge, 2023-11-20:
Timeout is now correctly implemented for `connect()`.

PR #12923 (2023-11-20): esp32: Implement socket connect timeout.
LWIP doesn't implement a timeout for blocking connect(), and such a timeout is not required by POSIX. However, CPython will use the socket timeout for blocking connect on most platforms. The "principle of least surprise" suggest we should support it on ESP32 as well (not to mention it's useful!)

This provides the additional improvement that external exceptions (like KeyboardInterrupt) are now handled immediately if they happen during connect(). (i.e. Previously Ctrl-C would not terminate a blocking connect until connect() returned, but now it will.)

C [... cut]
closed 2023-11-20 by dpgeorge