micropython/micropython #8328: Modbus Module for ESP32

eydam-prototyping, 2022-02-19:
Hey, 

I created a Modbus Module for the ESP32 and need some people to test it :)
You can create a RTU-Master/Slave and TCP-Server/Client. 
My own tests are passing, but it would be great, if someone else could do some more tests.

You can find the code here:
https://github.com/eydam-prototyping/micropython/tree/TEy-Modbus-2022-01-09

Some pre compiled images and the documentation here:
https://drive.google.com/drive/folders/12ZPlzrywi3Wp9dnlOKTvR8OETy1F6aCX
(always choose the newest ones)

There are some known bugs:
https://github.com/eydam-prototyping/micropython/issues?q=is%3Aissue+is%3Aopen+label%3Amodbus+
If you know how to solve them, please let me know.

Maybe we could integrate it to the main repo at some point.

Best regards 
Tobias

beyonlo, 2022-02-21:
@eydam-prototyping 

That is a great work and will help so much the people. Modbus is essential protocol for the industry.

@dpgeorge is possible to have this module in the master, to maybe, have it by default in the next Micropython release for ESP32?

Thank you very much.

e135193, 2022-06-14:
@eydam-prototyping Can you please share an example usage python file so that people can test it.

I wish to test it as well.

Best Regards

phildealm, 2022-08-22:
Hi, 
any news on it?
I'm willing to test it as I do have a project that needs communicating using modbus RTU.

However, I was not able to get started as I lack some basic information on how to use it. 

thanks,

eydam-prototyping, 2022-08-23:
Hey @e135193  and @phildealm.

sorry for the late replay.
There is an example in the documentation. You can find it here:
https://drive.google.com/drive/folders/12ZPlzrywi3Wp9dnlOKTvR8OETy1F6aCX (the PDF-File).

Is that enough? Or do you need more?

Best regards,
Tobias

beyonlo, 2022-10-03:
Hello @eydam-prototyping 

I would like to use that `ModBus` module on the `ESP32-S3,` over `WiFi` and `485.` Could you to build supporting the `ESP32-S3` as well? Or are there a doc where I can to build by myself the `ModBus` module for other `ESP32` chip, like as the `S3?`

Thank you very much!