micropython/micropython #8330: Socket datas (UDP, TCP, MQTT) are received discontinuously - expected real-time

Koraze, 2022-02-20:
Hi, for games and robotics projects, I send data from my computer to my ESP32 over wifi every 100ms (for TCP / MQTT and 50ms for UDP). The ESP32 shows me all the data sent (with no lost with TCP / MQTT and almost no lost with UDP). So the ESP32 correcty receives all data sent from my computer. The data payload is : `{"x": 1, "y": 2, "r": 102, "g": 0, "b": 0}`. 

Sadly, the ESP32 doesn't show me the data in real-time. It shows me up to 5 datas in a row instead of one data every 100ms. Results : the actions performed by the ESP32 are not fluid, like a lag in a video game. And I found no solutions on the internet.

I checked if the issue comes from my computer, its not. Other devices (other computers and smartphones in my local network) correctly shows me datas in real-time (so every 100ms with TCP / MQTT, or 50ms with UDP). I also tried to program my ESP32 with Arduino IDE and C++, no problem at all. So the issue seems to be related with MicroPython.

Could you help me to resolve this issue ? You can see below the simpliest UDP code I use for my tests.

```python
import network
import socket
import time

port_dest = 6000
ip_dest = "0.0.0.0"

session=socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
session.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)

try:
    session.bind((ip_dest, port_dest))
    print('waiting...')
    while True:
        data,addr=session.recvfrom(128)
        print('received:',data,'from',addr)
        print('sending:',data,'from',addr)
except :
    print("quit")
    session.close()
```

(btw now I can receive UDP data inside a thread for hours without crashes, it wasn't possible some years ago. Great thanks for your hard work)

dpgeorge, 2022-02-21:
Thanks for the report.

The issue could be related to WiFi power management.  When I ping an esp32 (which does not go through any MicroPython code, it's all in the IDF/lwip) the latency is up to 200ms and averages around 100ms.

Can you try ping'ing your esp32 when running MicroPython (no code needs to be running, just a REPL and connected to the WiFi), and then compare that to ping'ing it running the Arduino C++ code?

Koraze, 2022-02-21:
I try 12 pings for both, you can read here the average ping :
- With micropython wifi repl : 150,75 ms
- With arduino IDE C++ : 143 ms

So I'm stupid for reporting a false issue :(
I don't know why, but my old C++ udp code I wrote one or two years ago seemed to work smooter ... 

Sorry for reporting an false issue
Have a good day

closed 2022-02-21 by Koraze

dpgeorge, 2022-02-21:
> I don't know why, but my old C++ udp code I wrote one or two years ago seemed to work smooter ...

Do you still have that source code and can recompile it and test it today?

It may be possible to disable WiFi power saving to have a much smaller ping latency.

Koraze, 2022-02-21:
I didn't find it. I will let you know if I find it.
Hopefully, you add ESP-NOW in micropython. I heard the latency is between 30-50ms (at least with C++ codes).
Sadly ESPs can't communicate for now with PC or Smartphone with such this protocol, but I will try it :)
I will give you news soon

Koraze, 2022-02-21:
> I didn't find it. I will let you know if I find it. Hopefully, you add ESP-NOW in micropython. I heard the latency is between 30-50ms (at least with C++ codes). Sadly ESPs can't communicate for now with PC or Smartphone with such this protocol, but I will try it :) I will give you news soon

Koraze, 2022-02-21:
> > I don't know why, but my old C++ udp code I wrote one or two years ago seemed to work smooter ...
> 
> Do you still have that source code and can recompile it and test it today?
> 
> It may be possible to disable WiFi power saving to have a much smaller ping latency.

Do you know how to disable WiFi power saving in micropython ?

According to the link below, I tried to add _listen_interval_ parameter inside connect function in order to disable WiFi power saving, but only got an issue ...
https://micropython-tve.readthedocs.io/en/esp32-low-power/library/esp32-lowpower.html#esp32-specific-low-power-options-in-network

The code i tried (https://docs.micropython.org/en/latest/esp32/quickref.html#installing-micropython) : 
```python
def do_connect():
    import network
    sta_if = network.WLAN(network.STA_IF)
    if not sta_if.isconnected():
        print('connecting to network...')
        sta_if.active(True)
        sta_if.connect('<essid>', '<password>', listen_interval=-1)
        while not sta_if.isconnected():
            pass
    print('network config:', sta_if.ifconfig())
```

dpgeorge, 2022-02-21:
See #5473 for an old PR for configuring the `listen_interval` and wifi power saving.

Koraze, 2022-02-24:
PR files means "Pull Requests Files" ?
I don't really understand how to configuring the `listen_interval` thing ..