micropython/micropython #8332: rp2: Repl hangs when printing in scheduled generator

nekolab, 2022-02-20:
You can reproduce this bug constantly by using following code and please use a wire to shortcut GPIO 22 and 26

```python
from machine import SPI, Pin

from micropython import schedule

import micropython

i = Pin(22, Pin.IN, Pin.PULL_UP)
o = Pin(26, Pin.OUT, Pin.PULL_UP)

led = Pin(25, Pin.OUT, value = 0)

class IRQMachine:

    def __init__(self, irq_pin):
        self.irq_pin = irq_pin
        self.irq_queue = []
        self._irq_executor_ref = self._irq_executor
        self.irq_pin.irq(trigger=Pin.IRQ_FALLING | Pin.IRQ_RISING, handler=lambda pin: schedule(self._irq_executor_ref, pin.irq().flags()))

    def irq(self, func, *args):
        gen = func(*args)
        self.irq_queue.append(gen)
        next(gen)

    def _irq_executor(self, irq_flags):
        for gen in self.irq_queue:
            try:
                gen.send(irq_flags)
            except StopIteration:
                self.irq_queue.remove(gen)

def print_something():
    print("hello")
    flags = yield
    if flags & Pin.IRQ_FALLING:
        print('pull down')
    micropython.mem_info()

o.value(1)
machine = IRQMachine(i)
machine.irq(print_something)
o.value(0)
```

The code itself is a simple co-routine implementation based on generator, the method will stop execution on `yield` and restore execution when IRQ comes

After running the code, output will look like this, the `pull down` became unexpected red and REPL stops work
[image]

But if we remove print(), replace it with something like turn on the LED, it will work fine

Tested on Raspberry Pi Pico, not sure other ports/boards has same issue

peterhinch, 2022-03-09:
I can't replicate this - but see below. Pasting the code into mpremote it runs correctly:
```python
hello
>>> pull down
stack: 844 out of 7936
GC: total: 192064, used: 16352, free: 175712
 No. of 1-blocks: 309, 2-blocks: 25, max blk sz: 72, max free sz: 9985

>>> 
```
However note that a REPL prompt appears before the "pull down" text. This is presumably down to the timing of `micropython.schedule()`. My guess is that the REPL prompt is confusing whatever IDE you are using.

yjchun, 2022-03-10:
Same with @peterhinch . The editor of the author seems thonny which I don't have but above code works fine in pycharm and vscode as well as in mpremote.

peterhinch, 2022-03-11:
A couple of general points about the code sample. There seems to be little point in using **micropython.schedule** as **Pin.irq** uses soft interrupts by default. Also there is no reason to specify a pullup on an output pin.

nekolab, 2022-03-11:
Hi @peterhinch @yjchun 
Thanks for you help!
Yes I'm using Thonny as IDE and never try other before, I used to think the REPL in IDE is just some terminal work and must be mature enough to have bugs but I'm wrong. Now in mpremote everything works fine, leaves it a Thonny side bug, I will report to them later.

PS: You are right, the PULL_UP is not need, I just do the copy-paste and forget remove it. Since I'm really new to micropython, are there any doc points out that how soft and hard IRQ scheduled in micropython?

peterhinch, 2022-03-12:
See [this doc](https://docs.micropython.org/en/latest/reference/isr_rules.html).

nekolab, 2022-03-18:
Thanks~
Since this is a Thonny bug, I'm closing this issue now

closed 2022-03-18 by nekolab