micropython/micropython #8334: mpy-cross could NOT compile under Windows 11 using Cygwin

xidameng, 2022-02-21:
I am using Cygwin environment under Windows 11. When I was compiling for the `mpy-cross` executable file, it failed with the following log.

my settings:
OS: Win 11
Toolchain: gcc 11.2
Python: python 3.9.0

Under Windows10, there is no issue at all.

`/cygdrive/d/Work/MicroPython/micropython/mpy-cross
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
gcc: fatal error: no input files
compilation terminated.
gcc: fatal error: no input files
compilation terminated.
gcc: fatal error: no input files
gcc: fatal error: no input files
compilation terminated.
compilation terminated.
Traceback (most recent call last):
  File "D:\Work\MicroPython\micropython\py\makeqstrdefs.py", line 188, in <module>
    preprocess()
  File "D:\Work\MicroPython\micropython\py\makeqstrdefs.py", line 62, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "C:\Users\a0126\AppData\Local\Programs\Python\Python39\lib\multiprocessing\pool.py", line 870, in next
gcc: fatal error: no input files
compilation terminated.
    raise value
  File "C:\Users\a0126\AppData\Local\Programs\Python\Python39\lib\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "D:\Work\MicroPython\micropython\py\makeqstrdefs.py", line 46, in run
    return subprocess.check_output(args.pp + flags + files)
  File "C:\Users\a0126\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\a0126\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['gcc', '-E', '-I.', '-Ibuild', '-I..', '-Wall', '-Werror', '-Wextra', '-Wno-unused-parameter', '-Wpointer-arith', '-std=gnu99', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-Os', '-fdata-sections', '-ffunction-sections', '-fno-asynchronous-unwind-tables', '-DNO_QSTR', '../py/mpstate.c', '../py/malloc.c', '../py/gc.c', '../py/pystack.c', '../py/qstr.c', '../py/vstr.c', '../py/mpprint.c', '../py/unicode.c', '../py/mpz.c', '../py/reader.c', '../py/lexer.c', '../py/parse.c', '../py/scope.c', '../py/compile.c', '../py/emitcommon.c', '../py/emitbc.c', '../py/asmbase.c', '../py/asmx64.c', '../py/emitnx64.c']' returned non-zero exit status 1.
gcc: fatal error: no input files
compilation terminated.
gcc: fatal error: no input files
compilation terminated.
gcc: fatal error: no input files
compilation terminated.
make: *** [../py/mkrules.mk:105: build/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build/genhdr/qstr.i.last'
`

stinos, 2022-02-21:
> Under Windows10, there is no issue at all.

Was this an in-place upgrade from Window 10 and is everythng else (build tools version, Python version, Cygwin version, source code paths, ...) the same?

Does something like 

```
gcc -E -I. -Ibuild -I.. -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -std=gnu99 -DFFCONF_H="lib/oofatfs/ffconf.h" -Os -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables -DNO_QSTR ../py/mpstate.c
```

work when pasted in the shell while in the mpy-cross directory?

xidameng, 2022-02-22:
> Was this an in-place upgrade from Window 10 and is everythng else (build tools version, Python version, Cygwin version, source code paths, ...) the same?

nope, its a new PC with newly upgrade OS, so everything is new, Not a in-place upgrade

stinos, 2022-02-22:
But you used the same version of everything on Windows 10?
Did you get a chance to try the suggested command?

Also when you say 'toolchain gcc 11.2' does that mean you installed mingw64-x86_64-gcc  via Cygwin's setup.exe, or something else?

xidameng, 2022-02-28:
> But you used the same version of everything on Windows 10? Did you get a chance to try the suggested command?

Yes I did, and the bash returns what seems like a pre-processed file with no error thrown, the last few lines are as follows,

`# 28 "../py/mpstate.c" 2

mp_dynamic_compiler_t mp_dynamic_compiler = {0};

mp_state_ctx_t mp_state_ctx;
`

> Also when you say 'toolchain gcc 11.2' does that mean you installed mingw64-x86_64-gcc via Cygwin's setup.exe, or something else?

Yes and no, I installed the stock `gcc-core` from the Cygwin setup.exe, it's not the 64bit version. Here is its version information,

> $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/i686-pc-cygwin/11/lto-wrapper.exe
Target: i686-pc-cygwin

stinos, 2022-02-28:
I cannot reproduce this with these steps:
- fresh Win11 installation (10.0.22000 Build 22000, in a VM)
- download https://cygwin.com/setup-x86.exe
- run it and install latest of: git, make, gcc-core, python3.9

```
git clone https://github.com/micropython/micropython.git
cd micropython/mpy-cross
make
```

builds mpy-cross without errors. So does using `make -j4` instead (tested that to make sure it's not the multiprocess stuff).

Something on your setup is different. Not clear whether this is really a micropython issue rather than an environment one.

xidameng, 2022-02-28:
> I cannot reproduce this with these steps:
>
> * fresh Win11 installation (10.0.22000 Build 22000, in a VM)
> * download https://cygwin.com/setup-x86.exe
> * run it and install latest of: git, make, [...]

really appreciate your help and sharing, my laptop was recently upgraded to Win11, and I never use it for programming at all, so it should be quite "clean". My friends also used Win11 on VM and had the same issue.

I will try the `make -j4` command and see if it works for me

stinos, 2022-02-28:
That's unlikely to fix something. If anything you'd have to unisnatll Cygwin, install again with only the steps above then see if that works. I just remembered one crucial thing: I'm pretty sure the Python version used must be the one installed by Cygwin, not a CPython installation on Windows itself..

xidameng, 2022-02-28:
You are right, still same issue, regardless of which shell I used -- git bash, window prompt, powershell and cygwin bash, they all return the same error, as if gcc couldn't find the input files...

stinos, 2022-02-28:
Note I edited my comment right after posting, did you use the Python version which comes with Cygwin? From your first post it looks like you're using a Windows install (C:\Users\a0126\AppData\Local\Programs\Python\Python39).

xidameng, 2022-04-01:
> From your first post it looks like you're using a Windows install (C:\Users\a0126\AppData\Local\Programs\Python\Python39).

How to disable this one and use the Python I need?

stinos, 2022-04-01:
Normally that works automatically if you install Python in Cygwin: it will be installed in /usr/bin or similar, and in the `PATH` environment variable '/usr/bin` comes before any Windows paths so the python command will be picked up from /usr/bin.

xidameng, 2022-04-01:
> Normally that works automatically if you install Python in Cygwin: it will be installed in /usr/bin or similar, and in the `PATH` environment variable '/usr/bin` comes before any Windows paths so the python command will be picked up from /usr/bin.

Thanks for your reply, I did, I have already installed Python 3 using Cygwin, when I type `python3` in Cygwin bash, the python running is also from Cygwin dir instead of Windows one. 

I have tried multiple times, and the results are the same

stinos, 2022-04-01:
> when I type python3

And if you type `python`?

> I have tried multiple times, and the results are the same

Are you saying that the error still contains lines like `"C:\Users\a0126\AppData\Local\Programs\Python\Python39\lib\multiprocessing\pool.py"`? I.e. from the Windows installation?

What's the output of `make V=1 PYTHON=python3` ?

xidameng, 2022-04-01:
> And if you type `python`?

Its the same thing, same as when typing python3

jonnor, 2024-09-08:
Were you able to resolve your problem @xidameng  ?

So far it seems that others have not been able to reproduce the issue.

jonnor, 2025-03-03:
Not reproducible.

closed 2025-03-03 by jonnor