micropython/micropython #8336: STM32H7: PA0 and PA1 ADC returning incorrect values

EdgarTheEE, 2022-02-22:
Hello community,
I have a custom stm32H7 board running v1.17 of micropython.
I have noticed that PA0 and PA1 return incorrect ADC values:

```
from machine import ADC, Pin
adc = ADC(Pin.cpu.A0)
adc.read_u16()
```
The return value is 5763 (we are expecting ~61722 with a 3.3Vref) the first time "adc.read_u16()" is called. Calling "adc.read_u16()" will then consistently return ~800 after the first call. PA3 and PA4 are working as expected. Are these two pins reserved for a specific function?

EdgarTheEE, 2022-02-22:
Looks like I fixed the issue.
I edited stm32h743_af.csv:
Removed "ADC12_INP0" from PA0 AF
Removed "ADC12_INP1" from PA1 AF
Not entirely sure why this fixes the issue.

While doing so, I also discovered two other related issues in stm32h743_af.csv:
PF12 ADC section (line 95 has a typo):
`ADC1_INN2/DC1_INP6`
This should be ADC1_INN2/**A**DC1_INP6

PF13 ADC section (line 96 has a typo):
`ACD2_INP2`
This should be A**DC**2_INP2

dpgeorge, 2022-02-22:
PA0 can be on ADC12_INP0 and ADC1_INP16.  So if ADC1 is selected (which is the default) then I'm not sure how it distinguishes/selects between INP0 and PIN16 (channel 0 vs channel 16).  That's probably the root of the issue.  Removing ADC12_INP0 from the af.csv file makes it work because now it's configuring ADC1 channel 16.