micropython/micropython #8338: msbuild error MSB4041 on VS2015

dotnfc, 2022-02-22:
i tried vs2015 to build the latest mpy v1.8, failed with error MSB4041:

D:\Work\Temp\mpy-1v8a\mpy-cross>msbuild /p:Configuration=Release mpy-cross.vcxproj
D:\Work\Temp\mpy-1v8a\ports\windows\variants\standard\mpconfigvariant.props(1,1): error MSB4041: ... ...

fixed by modifiying the {mpy}\ports\windows\variants\standard\mpconfigvariant.props first line to

`
<?xml version="1.0" encoding="utf-8"?>
<Project DefaultTargets="Build" ToolsVersion="14.0" xmlns="http://schemas.microsoft.com/developer/msbuild/2003">
`

stinos, 2022-02-22:
Can you try to figure out which of these is really needed? E.g. does

```
<?xml version="1.0" encoding="utf-8"?>
<Project xmlns="http://schemas.microsoft.com/developer/msbuild/2003">
```

also work? Or perhaps even

```
<?xml version="1.0" encoding="utf-8"?>
<Project">
```

dotnfc, 2022-02-22:
for vs2015, msbuild.exe may check the xmlns="" item, but vs2019, vs2022 not. ( i have not tested the vs2017 ). 
the MSB4041  error full message looks like

```
error MSB4041: The default XML namespace of the project must be the MSBuild
 XML namespace. If the project is authored in the MSBuild 2003 format, please add xmlns="http://schemas.microsoft.com/d
eveloper/msbuild/2003" to the <Project> element. If the project has been authored in the old 1.0 or 1.2 format, please
convert it to MSBuild 2003 format
```

so, if there is no xmlns='', the build would failed on vs2015.

stinos, 2022-02-22:
Ok thanks, PR with fix #8339

dpgeorge, 2022-02-25:
Fixed by c2e8a5acd267e55a11a709d44f0385d0a40f3ec9

closed 2022-02-25 by dpgeorge