micropython/micropython #8340: Using asyncio in _thread

yjchun, 2022-02-22:
Hello.
This might be a feature request: asyncio is missing support for threading

```python
import uasyncio as asyncio
import _thread
import time

async def test1(arg):
    try:
        while True:
            print(arg)
            await asyncio.sleep(1)

    except Exception as e:
        print('Exception in '+arg)
        print(e)

def run(arg):
    try:
        asyncio.run(test1(arg))
    except Exception as e:
        print('Exception in asyncio.run ' + arg)
        print(e)
    print("END "+arg)

_thread.start_new_thread(run, ('core1',))
time.sleep_ms(100)
run('core0')
```

Tested in two ports: unix with uasyncio module included and rp2 with https://github.com/micropython/micropython/pull/8310 patch
Exception within a few seconds in both ports.
Sometimes `ValueError: generator already executing`
Sometimes `'NoneType' object has no attribute 'ph_child'`

Also I would like to create new event loop in new thread (which is automatically done in CPython). Do you think it is possible to implement proper event loop? or would there be limitations in VM? (is generator is thread safe in VM level?)

peterhinch, 2022-02-23:
You beat me to it. Give #8310 and apparently reliable two-core operation I was about to try this. I can see a definite use case for a copy of `uasyncio` running on each core.

dpgeorge, 2022-02-25:
Yes this would be a good enhancement.

(Related, I've sometimes wanted to run a separate, independent uasyncio instance within a soft IRQ handler...).

dpgeorge, 2022-03-29:
Some thoughts on this:
- all global state in `uasyncio` (which is mostly in `core.py`) needs to be duplicated for each core/thread
- then there needs to be a way for all `uasyncio` functions to know what state they are associated with
- it would make sense to put all state in `Loop`, and then it might be possible to pass a `loop` argument around so that functions know the asyncio instance they are attached to.  But this would require passing `loop` into everything, including calls like `.read()` and `.write()`, and that would break all existing asyncio based code
- instead of passing `loop` around, it would need to be made implicit and the current `loop` instance instead stored in some thread-local-state
- for running `uasyncio` instances within a soft interrupt handler, that code can explicitly set the `loop` instance on entry to the handler, then restore the original `loop` on exit