micropython/micropython #8346: Neopixel appears broken in v1.18 on esp32-c3

jcksnvllxr80, 2022-02-24:
I was running micropython v1.17 and neopixel was working perfect to control the onboard LED using pin 8.
I upgraded to v1.18 and all of a sudden, the write() function on NeoPixel doesn't even seem to work using the example code in the docs. (granted I am using the onboard LED instead of external neopixels). Anyone know whats going on here? Error shown below.

```
from machine import Pin
from neopixel import NeoPixel
pin = Pin(8, Pin.OUT)
np = NeoPixel(pin, 8)
np[0] = (255, 255, 255)

np.write()
  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    File "neopixel.py", line 1, in write
  OSError: (-258, 'ESP_ERR_INVALID_ARG')
```

dpgeorge, 2022-02-24:
Please see related #8198.

I think the bug here was fixed by 1f04a9a1fcd4172f7c721e05a4515c9c8bb0e8c9

jcksnvllxr80, 2022-02-25:
I would like to fix this issue by using the latest nightly build [here](https://micropython.org/resources/firmware/esp32c3-20220224-unstable-v1.18-149-g64bfaae7a.bin) but in its name it says "unstable". I am not exactly sure that means what I think it means. Does it mean, "do not install this because it is doomed to break"?

dpgeorge, 2022-02-25:
Nightly/unstable builds are relatively OK.  They already pass a lot of tests to get to that stage.  But don't use it in mission critical things!

jcksnvllxr80, 2022-02-25:
Yes, the latest nightly build fixed the issue. Thank you

closed 2022-02-25 by jcksnvllxr80