micropython/micropython #8352: Error running microprython on esp32-s3

Mars-CN, 2022-02-25:
After recompiling microprython on esp32-s3, I can download it normally and run it normally.

But there was an error when starting
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0xe34
load:0x403b6000,len:0x9e8
load:0x403ba000,len:0x2be4
entry 0x403b61d4
Performing initial setup
Traceback (most recent call last):
File "_boot.py", line 11, in <module>
File "inisetup.py", line 36, in setup
OSError: 28
MicroPython v1.18 on 2022-02-15; ESP32 module with ESP32
Type "help()" for more information.
>>>

The information returned by executing the command seems to be incorrect:
bdev : <Partition type=1, subtype=129, address=2097152, size=14000000, label=vfs, encrypted=0>
uos.statvfs("") : (0, 0, 0, 0, 0, 0, 0, 0, 0, 128)
uos.stat(""): (16384, 0, 0, 0, 0, 0, 0, 0, 0, 0)

The printing information mode in the startup parameters is slow_ Rd, but I have set it to DIO mode in menuconfig
What's wrong?

board : ESP32-S3-DevKitC-1
module : D3N32R8V

jupitergao18, 2022-06-17:
module : D2N16R8V

0xabe1, 2022-11-02:
This is still a problem. Just encountered it.

To unblock myself, I did notice that loading UF2 and CircuitPython properly formats the file system. Then you can reload uPy (double click Reset and load the uPy .uf2 file) and it boots correctly. This appears to be a filesystem formatting problem on first boot.

jonnor, 2024-09-08:
I have used MicroPython 1.22 and later on multiple ESP32-S3 devices, without encountering this problem.
Note that it may be needed to use erase_flash to get rid of any prior incompatible second-stage bootloaders, filesystems, etc. And make sure to get the program offset correctly for the flash command.

Can you re-test if erasing flash and flashing latest MicroPython (1.23 or later), fixes the problem for you?

jonnor, 2025-03-03:
Not reproducible.