micropython/micropython #8353: Error building firmware

fanrenkong, 2022-02-25:
I only recently started learning micropython and was lucky enough to discover the [ulab](https://github.com/v923z/micropython-ulab) module. I followed the steps exactly to build the firmware for my esp32 board, but when I in the last step, I got the following error:

/Volumes/CODE/ESP32/micropython/micropython/ports/esp32/modmachine.c:40:10: fatal error: esp32/clk.h: No such file or directory
#include "esp32/clk.h"
^~~~~~~~~~~~~
compilation terminated.
/Volumes/CODE/ESP32/micropython/micropython/ports/esp32/main.c:38:10: fatal error: soc/cpu.h: No such file or directory
#include "soc/cpu.h"
^~~~~~~~~~~
compilation terminated.
/Volumes/CODE/ESP32/micropython/micropython/ports/esp32/gccollect.c:37:10: fatal error: soc/cpu.h: No such file or directory
#include "soc/cpu.h"
^~~~~~~~~~~
compilation terminated.

I have tried several times and don't know what is the problem, can you give me some help?

robert-hh, 2022-02-25:
You must have missed some step in setting up the build environment. In the esp-idf directory, is it essential to update the submodules, like:

git submodule sync
git submodule update --init

fanrenkong, 2022-02-25:
Thanks for you suggestion, I figured out that the problem should occur when installing esp-idf, by following the instructions carefully again, I successfully compiled the firmware.

closed 2022-02-25 by fanrenkong