micropython/micropython #8355: ESP 32 OTA with v1.18

VarshneyNaupada, 2022-02-25:
Hi Team,

I have flashed " esp32-ota-20220117-v1.18.bin " to my esp32 with 4MB flash device, After flashing it it took almost 1.5 MB for firmware partition and rest for others as per given below partition table.

As per below table and v1.18 bin file size, there is no possibility for OTA updates with provision of having factory , OTA_0 & OTA_1, along with application code. please correct me if my understanding is not valid.

Also, I did not find any sample code for implementation of OTA along with application code . Requesting team to include sample files or guide us the implementation instructions with this version.

# Partition table for MicroPython with OTA support using 4MB flash
# Notes: the offset of the partition table itself is set in
# $IDF_PATH/components/partition_table/Kconfig.projbuild.
# Name,   Type, SubType, Offset,   Size,     Flags
nvs,      data, nvs,     0x9000,   0x4000,
otadata,  data, ota,     0xd000,   0x2000,
phy_init, data, phy,     0xf000,   0x1000,
ota_0,    app,  ota_0,   0x10000,  0x180000,
ota_1,    app,  ota_1,   0x190000, 0x180000,
vfs,      data, fat,     0x310000, 0x0f0000,

Thank you

VarshneyNaupada, 2022-03-01:
@dpgeorge : Hi George, could you please suggest me any update or approach on this thread

AmirHmZz, 2022-03-11:
If you flash the `ota` version, you can see the firmware is flashed on `ota_0` so I think it is possible anyway!

```Python
>>> from esp32 import Partition
>>> Partition(Partition.RUNNING)
<Partition type=0, subtype=16, address=65536, size=1572864, label=ota_0, encrypted=0>
```

VarshneyNaupada, 2022-03-14:
Hi @AmirHmZz , 

I have mentioned flash is successful, but if you look at the size of flash it is 1.5 MB size without any application code. So usually ESP32 comes with 4MB only custom modules comes with higher flash memory.

If ota_0 takes 1.5MB without application code. It is difficult to do ota updates.

Usually we should have Factory partition, ota_0 & ota_1 partitions to have complete ota features.

I am looking to get some help to minimise the ota firmware file size to some where 1MB with application code.

If you have any suggestions for implementation let us know. 

Thank you !!

AmirHmZz, 2022-03-15:
@VarshneyNaupada I couldn't understand why do you need `factory` partition when you have `ota_0` and `ota_1` partitions?
Anyway, I want to share my partition table in case you want to maximize size of `ota` partitions and minimize the `vfs` partition :
```CSV
# Name,   Type, SubType, Offset,   Size,     Flags
nvs,      data, nvs,     0x9000,   0x4000,
otadata,  data, ota,     0xd000,   0x2000,
phy_init, data, phy,     0xf000,   0x1000,
ota_0,    app,  ota_0,   0x10000,  0x1e0000,
ota_1,    app,  ota_1,   0x1f0000, 0x1e0000,
vfs,      data, fat,     0x3d0000, 0x30000,
```

VarshneyNaupada, 2022-03-16:
Hi @AmirHmZz ,
I understand your point if we have 2 Ota partitions we can switch to either of them. 

Thanks for providing your partition table. I need your help to under stand or any link which can help me to create custom build with application code.

I wanted to minimise the OTA and give some additional size to data(vfs) partition to store some default values and running data. Is there any way to do it ?

AmirHmZz, 2022-03-16:
@VarshneyNaupada for that case use this `[partition table](https://github.com/micropython/micropython/blob/master/ports/esp32/partitions-ota.csv)`. It has maximized size of `vfs` partition.

BetterAutomations, 2022-04-11:
FYI I was able to implement OTA by heavily modifying tve's code:
https://github.com/tve/mqboard/blob/master/mqrepl/mqrepl.py#L22-L84

deus-est-machina, 2023-03-24:
@BetterAutomations 

Would you mind sharing your heavily modified code?

BetterAutomations, 2023-03-24:
> @BetterAutomations
> 
> Would you mind sharing your heavily modified code?

Don't want it publicly released but if I can get it to you some other way I will

deus-est-machina, 2023-03-26:
@BetterAutomations private repo and invite?

BetterAutomations, 2023-03-27:
> @BetterAutomations private repo and invite?

I've added you to the repository. It's sloppy and not the best quality, and it even may no longer work, but here it is :-) I do need to ensure it's working sometime in the next six months so if you get stuck with it just ping me and I'll let you know if I've got an updated, better-working version.

deus-est-machina, 2023-03-28:
Thanks.

I'll have a look ^^

jonnor, 2024-09-08:
Does anyone have a guide on how to do OTA on ESP32 with recent MicroPython?

jonnor, 2024-09-08:
Related: https://github.com/micropython/micropython/issues/8011

BetterAutomations, 2024-10-17:
> Does anyone have a guide on how to do OTA on ESP32 with recent MicroPython?

I just found out about this:
https://github.com/glenn20/micropython-esp32-ota

And see also this for manipulating the image for OTA without a full recompile:
https://github.com/glenn20/mp-image-tool-esp32

There's also this older project, which may or may not work with newer releases:
https://github.com/rdehuyss/micropython-ota-updater

Then there's my code; If you want to see it, I can add you. It's pretty sloppy, but it was working as of uPy v1.16. Have not tested it yet on newer.