micropython/micropython #8356: mpy-cross: Build process broken.

robert-hh, 2022-02-25:
Since the update of mpy-cross with v1.18-155 I get compile errors in the generated frozen-content.c file. See the dump below. The file contains indeed syntax errors. The build target does not matter. mpy-cross is fresh built.
The file that cannot be frozen is pye.py https://github.com/robert-hh/Micropython-Editor/blob/master/pye.py
It contains some escape sequences. A short excerpt from this file which provokes an error:
```
def get_screen_size(self):
    self.wr('\x1b[999;999H\x1b[6n')
    pos = ''
    char = self.rd()
    while char != 'R':
        pos += char
        char = self.rd()
    return [int(i, 10) for i in pos.lstrip("\n\x1b[").split(';')]
```
```
build-TEENSY40/frozen_content.c:5278:1: error: stray '\33' in program
 5278 | [
      | ^
build-TEENSY40/frozen_content.c:5279:9: error: expected ']' before ',' token
 5279 |     0x36,0x01, // CALL_METHOD 1
      |         ^
      |         ]
build-TEENSY40/frozen_content.c:5285:1: error: expected expression before '}' token
 5285 | };
      | ^
build-TEENSY40/frozen_content.c:5309:17: error: 'fun_data_tilt_over__lt_listcomp_gt_' undeclared here (not in a function); did you mean 'raw_code_tilt_over__lt_listcomp_gt_'?
 5309 |     .fun_data = fun_data_tilt_over__lt_listcomp_gt_,
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 raw_code_tilt_over__lt_listcomp_gt_
make: *** [../../py/mkrules.mk:77: build-TEENSY40/build-TEENSY40/frozen_content.o] Fehler 1

```

yjchun, 2022-02-25:
I am also having problem with lastest master, fresh built. Corresponding frozen-content.c locations are:
```
    0x10,0x23, // LOAD_CONST_STRING /
    0x10,0x24, // LOAD_CONST_STRING \
```
```
    0xc5, // STORE_FAST 5
    0x10,0x81,0x1a, // LOAD_CONST_STRING {}
{{{}}}
    0x14,0x7d, // LOAD_METHOD format
```
And there is a related post in the forum: https://forum.micropython.org/viewtopic.php?f=21&t=12058

robert-hh, 2022-02-25:
Thanks. I recall having read about that before, but could not find it under the issues in the repository. The change suggested by @hippy works for me as well.

closed 2022-02-28 by robert-hh