micropython/micropython #8358: pyboard.py very slow on windows

harbaum, 2022-02-26:
I have recently adopted my µPIDE Micropython IDE to use pyboard.py for remote communication. I soon noticed that transfers from the device to the host are 5 times slower under Windows 10 than under Linux. My previous own version did not have the issue,

It turned out that serial.inWaiting() is awfully slow on Windows 10 and it's being called for every single byte that's being read via read_unit(). 

As a solution I wrote a little buffering serial wrapper.
  https://github.com/harbaum/upide/blob/main/src/buffered_serial.py

This can be used in pyboard.py like `import buffered_serial as serial` in https://github.com/micropython/micropython/blob/0a217624e1d26899986d3e79eb604652da540b00/tools/pyboard.py#L268

This avoids calling inWaiting() permanently and as a result pyboard.py's read performance now feels similar to Linux.

harbaum, 2022-02-28:
Some more infos: I see this with various ESP32 boards using the CH340 and the CP2102 USB bridges. So this is not related to a certain USB driver. However, I have not tested with a CDC device. But the buffering won't hurt there.

harbaum, 2022-03-05:
The buffer works fine the LEGO SPIKE and thus also works with CDC devices.

jonnor, 2024-09-08:
mpremote is now preferred over pyboard. Is this relevant also for mpremote?

Josverl, 2024-09-08:
@jonnor , It might , [parts of pyboard.py have been used in](https://github.com/micropython/micropython/commit/b4d785fa201bb2e9ecf5abb47e0cba21a9d8b6f7) `mpremote's transport_serial`