micropython/micropython #8360: rp2: UART TX can issue double characters

peterhinch, 2022-02-27:
This arises rarely when processes running on each core asynchronously write to the UART.

Test setup: two Picos one acting as transmitter, the other as receiver. The rx checks data, issues a pulse on GPIO2 if a good line is received. Sets GPIO3 high on failure, triggering the LA and terminating the transmitter script.

Time to occur is random but the outcome of multiple runs as seen on the LA is consistent.

 1. Fault occurs only when core 1 issues `.write` while the UART is still outputting data from a prior core 0 write. Timing is consistent: the fault occurs when the core 1 write occurs 22.7-22.9ms after the core 0 write.
 2. Though a lock is used to protect `.write` the problem is not due to a lock conflict: it occurs long after the lock is released.
 3. Never occurs when core 0 issues `.write` while a core 1 transmission is in progress.
 4. Never occurs when a write is issued and no transmission is in progress.
 5. When the fault occurs UART issues a burst of about 1-15 double characters and then recovers.
 6. Time to fail is variable from seconds to ~30 minutes.

In the following, traces are as follows:
 1. UART data (note burst of doubles from "ee" to "pp").
 2. FAIL signal from receiver (LA trigger occurs off-screen to right).
 3. High if core 0 is holding the lock (duration of `.write`).
 4. High if core 1 is holding the lock (duration of `.write`).
 5. Pulse when receiver has seen a valid line.

[image]

The transmit code is as follows:
```python
from machine import UART, Pin
import _thread
from time import sleep

lock = _thread.allocate_lock()
uart = UART(0, 9600)
p3 = Pin(3, Pin.IN)
p4 = Pin(4, Pin.OUT)
p5 = Pin(5, Pin.OUT)

msg = (b"ABCDEFGHIJKLMNOPQRTSUVWXYZ\n", b"abcdefghijklmnopqrstuvwxyz\n")

def other():
    while not p3():
        lock.acquire()
        p5(1)
        uart.write(msg[1])
        lock.release()
        p5(0)
        sleep(0.105)  # Note rep rate of other thread
    
_thread.start_new_thread(other, ())
while not p3():
    lock.acquire()
    p4(1)
    uart.write(msg[0])
    lock.release()
    p4(0)
    sleep(0.1)

print("Terminated by receiver.")
```
The receiver is:
```python
from machine import UART, Pin

uart = UART(0, 9600, timeout=1000)
p2 = Pin(2, Pin.OUT, value=0)
p3 = Pin(3, Pin.OUT, value=0)

msg = (b"ABCDEFGHIJKLMNOPQRTSUVWXYZ\n", b"abcdefghijklmnopqrstuvwxyz\n")
print("Waiting")
def getline():
    while not uart.any():
        pass
    res = uart.readline()
    print(res)
    return res in msg

while not getline():
    pass
print("Started")
while getline():
    p2(1)
    p2(0)
p3(1)
print("Fail")
```

yjchun, 2022-02-28:
hello @peterhinch 
Can you test with following patch?
This probably fixes race condition with ISR.

```c
diff --git a/ports/rp2/machine_uart.c b/ports/rp2/machine_uart.c
index 0642f704f..3668539ba 100644
--- a/ports/rp2/machine_uart.c
+++ b/ports/rp2/machine_uart.c
@@ -456,7 +456,9 @@ STATIC mp_uint_t machine_uart_read(mp_obj_t self_in, void *buf_in, mp_uint_t siz
             }
             MICROPY_EVENT_POLL_HOOK
         }
+        self->read_lock = true;
         *dest++ = ringbuf_get(&(self->read_buffer));
+        self->read_lock = false;
         t = time_us_64() + timeout_char_us;
     }
     return size;
@@ -469,6 +471,7 @@ STATIC mp_uint_t machine_uart_write(mp_obj_t self_in, const void *buf_in, mp_uin
     const uint8_t *src = buf_in;
     size_t i = 0;
 
+    self->write_lock = true;
     // Put as many bytes as possible into the transmit buffer.
     while (i < size && ringbuf_free(&(self->write_buffer)) > 0) {
         ringbuf_put(&(self->write_buffer), *src++);
@@ -476,7 +479,6 @@ STATIC mp_uint_t machine_uart_write(mp_obj_t self_in, const void *buf_in, mp_uin
     }
 
     // Kickstart the UART transmit.
-    self->write_lock = true;
     uart_fill_tx_fifo(self);
     self->write_lock = false;
 
@@ -494,10 +496,10 @@ STATIC mp_uint_t machine_uart_write(mp_obj_t self_in, const void *buf_in, mp_uin
             }
             MICROPY_EVENT_POLL_HOOK
         }
+        self->write_lock = true;
         ringbuf_put(&(self->write_buffer), *src++);
         ++i;
         t = time_us_64() + timeout_char_us;
-        self->write_lock = true;
         uart_fill_tx_fifo(self);
         self->write_lock = false;
     }
```

peterhinch, 2022-02-28:
I am looking at something similar. This patch to **machine_uart_write** has run for a couple of hours:
```C
    // Kickstart the UART transmit. If core 1 issues write while a core 0 write already updating fifo, get dupes
    if (!self->write_lock) {
        self->write_lock = true;
        uart_fill_tx_fifo(self);
        self->write_lock = false;
    }
```
Prior to this I never managed more than 30 minutes.

However I think a mutex is the bomb-proof solution and I'm trying to figure out doing this.

dpgeorge, 2022-03-24:
I could finally reproduce this issue.  I had a Pico connected to a PYBD-SF2, with the Pico being the transmitter (of course).

Output on the SF2:
```
b'ABCDEFGHIJKLMNOPQRTSUVWXYZ\n'
b'abcdefghijklmnopqrstuvwxyz\n'
b'ABCDEFGHIJKLMNOPQRTSUVWXYZ\n'
b'abcdefghhiijjkkllmmnnooppqqrstuvwxyz\n'
Fail
```
I left it running and when I looked at 1hr 18 mins it had failed in this way.

closed 2022-04-12 by dpgeorge