micropython/micropython #8362: ESP32 / UART.deinit()

cederom, 2022-02-27:
Hello world :-)

I am using UART on ESP32 with MP 1.18. There is either bug in the documentation or ESP32 implementation of `UART.deinit()`.

I cannot call UART.deinit() just do "turn off UART" to be used for later because it seems to destroy object that cannot be then initialised again.

The scenario is:
* I have two UART based sensors. Both using different baudrate.
* Device should be able to work with both sensors either-or-exclusively or both of them based on additional pin activation.
* Both sensors can be attached to the same UART (only one sensor is active at time). They may also use different pins but the same UART (UART object may be reconfigured before read from each sensor).
* I have created a driver that initializes the UART with `UART.init()` according to the sensor needs.
* In order to conserve power each sensor is put to sleep by default, then wake up by pin and UART sequence, measurement is made, then sensor is put back to sleep with UART sequence and external pin.
* When there is a `UART.deinit()` call put in sensor deactivation call, another init/read/write to the UART object is impossible.

According to UART documentation [1] `UART.deinit()` only "Turn off the UART bus." but there is no mention that object is destroyed and cannot be used again. I would like to `init()` and `deinit()` the same UART bus depending on which sensor is being used.

My questions:
* Can I call `UART.init()` multiple times on the same object to change baudrate maybe even pins?
* Can I call `UART.deinit()` before `UART.init()` is called again on the same object?

Depending on the above it seem that either or:
* `UART.deinit()` destroys (not only turns off) the UART object so it cannot be used back again. The documentation needs to be updated in that case.
* There is a bug in ESP32 implementation that prevents further `UART.init()` after `UART.deinit()`.

[1] https://docs.micropython.org/en/latest/library/machine.UART.html?highlight=uart#machine.UART.deinit

robert-hh, 2022-02-27:
Most configuration of the UART constructor happens in code, that is shared with the init() call. So:

Yes, you can call uart.init() multiple times with different settings.
No, you cannot call uart.init() after uart.deinit(). After uart.deinit(), you have to create a new UART instance.

cederom, 2022-02-27:
Thank you @robert-hh :-) In that case Documentation needs an update, because "turn off UART" is not precise enough and does not imply destroying the object - I understood it only turns off the peripheral for saving power but it can be used later on with `init()` again :-)

robert-hh, 2022-02-27:
The state of an object after deinit() varies between classes and ports. So it is safe to assume, that after deinit() the respective object cannot be used anymore. In any case, a suitable object can be created again with little overhead, since most of the initialization is for most classes done by a function shared with init(), called init_helper().

cederom, 2022-02-27:
I have more complex program structure :-) There is a hardware definition class that holds the peripherals for a target variant of the hardware, so one firmware can be used on various devices :-) This hardware definition class initializes the peripherals and provide them to the drivers. Drivers can share one peripheral (i.e. UART) and can setup parameters at runtime. So peripheral re-creation is not really desired here when program is already running. In case of hardware failure error log backtrace is saved for later upload to the internet platform and device is rebooted :-)

Safest assumption is that `UART.deinit()` destroys peripheral and it needs to be re-created. If someone needs to re-configure the peripheral just another call to `UART.init()` is the best way. Thank you @robert-hh :-)

jonnor, 2024-09-08:
The question seems to have been answered. Proposing to close this.

closed 2024-09-08 by cederom