micropython/micropython #8366: Seemingly slow capturing of int in closures

stinos, 2022-02-28:
Originally reported here: https://forum.micropython.org/viewtopic.php?f=2&t=12060; 'seemingly' because I'm not sure what exactly is slow. Reproduce with:

```
import time

def ret(arg):
    yield None

def test(klass=int, count=10000):
    for n in range(count):
        ret(klass())

def measure(what, name):
    start = time.ticks_ms()
    for _ in range(10):
        what()
    end = time.ticks_ms()
    print(name, time.ticks_diff(end, start))

for typ in [int, float, dict, list]:
    measure(lambda: test(typ), typ)
```

Which yields numbers like this on windows and unix ports and reportedly does the same on RP2/pico:
```
<class 'int'> 1090
<class 'float'> 18
<class 'dict'> 14
<class 'list'> 19
```
Given the extreme difference this seems to indicate an underlying problem.

yjchun, 2022-02-28:
Please note that async function is a generator function too. Replace ret() with async_ret() and will see identical results.
``` 
async def async_ret(arg):
    pass
```

This problem is actually more relevant in asyncio.

dpgeorge, 2022-03-01:
It looks like this has to do with the GC doing excessive work to find some memory to create the generator (calling `ret`).  If `klass()` allocates memory then it's fast.  If it doesn't allocate memory then it's slow.

jonnor, 2024-09-08:
Is there anything more to investigate or fix here?

dpgeorge, 2024-09-09:
This is still a problem as of v1.24.0-preview.260.g7bb28f04e (tested unix and rp2).

It's not a bug but would be good to improve the GC to handle this case.