micropython/micropython #8375: FAILED: genhdr/qstr.i.last

nccwarp9, 2022-03-02:
Im trying to compile micropython for esp32 but Im stuck.

ncc@ncc-virtualbox:~/micropython/ports/esp32$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/ncc/micropython/ports/esp32/build
Executing "ninja all"...

**....**

[964/1193] Generating /home/ncc/micropython/mpy-cross/mpy-cross
make[1]: Entering directory '/home/ncc/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/mpversion.h
CC main.c
LINK mpy-cross
   text    data     bss     dec     hex filename
 304457   17184     840  322481   4ebb1 mpy-cross
make[1]: Leaving directory '/home/ncc/micropython/mpy-cross'
[965/1193] Generating ../../genhdr/qstr.i.last
**FAILED: genhdr/qstr.i.last** 

**....**

/home/ncc/micropython/ports/esp32/machine_sdcard.c
/home/ncc/micropython/ports/esp32/main.c:38:10: fatal error: soc/cpu.h: No such file or directory
 #include "soc/cpu.h"
          ^~~~~~~~~~~
compilation terminated.
/home/ncc/micropython/ports/esp32/gccollect.c:37:10: fatal error: soc/cpu.h: No such file or directory
 #include "soc/cpu.h"
          ^~~~~~~~~~~
compilation terminated.
In file included from /home/ncc/micropython/ports/esp32/machine_timer.c:38:
/home/ncc/esp/esp-idf/components/driver/deprecated/driver/timer.h:16:2: warning: #warning "legacy timer group driver is deprecated, please migrate to driver/gptimer.h" [-Wcpp]
 #warning "legacy timer group driver is deprecated, please migrate to driver/gptimer.h"
  ^~~~~~~
/home/ncc/micropython/ports/esp32/machine_timer.c:217:13: warning: 'TIMER_BASE_CLK' macro is deprecated
         self->period = (uint64_t)(TIMER_SCALE / mp_obj_get_float(args[ARG_freq].u_obj));
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ncc/micropython/ports/esp32/machine_timer.c:225:13: warning: 'TIMER_BASE_CLK' macro is deprecated
         self->period = (((uint64_t)args[ARG_period].u_int) * TIMER_SCALE) / args[ARG_tick_hz].u_int;
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ncc/micropython/ports/esp32/modmachine.c:40:10: fatal error: esp32/clk.h: No such file or directory
 #include "esp32/clk.h"
          ^~~~~~~~~~~~~
compilation terminated.

**...**

, '/home/ncc/micropython/ports/esp32/machine_sdcard.c']' returned non-zero exit status 1.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
ncc@ncc-virtualbox:~/micropython/ports/esp32$

dpgeorge, 2022-03-02:
- make sure you're on a valid IDF version (see esp32/README.md)
- make sure you run `git submodule update --init` in the IDF repo

nccwarp9, 2022-03-02:
IDF was installed following https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html
IDF is on version 4.4.
and git submodule update --init was run in /esp-idf folder

dpgeorge, 2022-03-02:
What is the output of the following commands in the IDF folder:
- `git status`
- `git describe`
- `find -name cpu.h`

pluzmedia, 2022-03-03:
Same problem here

`git status`
`On branch master`
`Your branch is up to date with 'origin/master'.`

`nothing to commit, working tree clean`

`git describe`
`v5.0-dev-1766-g84f9aa13c`

`find -name cpu.h`
`return nothing`

dpgeorge, 2022-03-03:
You need to be on tag `v4.4`, not latest master (of the IDF).

nccwarp9, 2022-03-03:
had the same output as pluzmedia

i did 
git checkout -b 4.4

:~/esp/esp-idf$ git status
On branch 4.4
nothing to commit, working tree clean

in micropython dir:
:micropython/ports/esp32$ . $HOME/esp/esp-idf/export.sh

:micropython/ports/esp32$ idf.py build

and i get this error 

[963/1191] Generating ../../genhdr/qstr.i.last
FAILED: genhdr/qstr.i.last 
...
micropython/ports/esp32/main.c:38:10: fatal error: soc/cpu.h: No such file or directory
 #include "soc/cpu.h"
          ^~~~~~~~~~~
compilation terminated.

robert-hh, 2022-03-03:
With `git checkout -b 4.4` you created a new branch of whatever branch you already were. The proper way would be:

``'
git checkout v4.4
git submodule sync
git submodudule update --init
```
and maybe:
./install.sh
to ensure, that the proper tools are loaded.

Then you can go on in the micropython directory.

nccwarp9, 2022-03-03:
~/esp/esp-idf$ git checkout 4.4
Already on '4.4'
ncc@ncc-virtualbox:~/esp/esp-idf$ git submodule sync
Synchronizing submodule url for 'components/asio/asio'
Synchronizing submodule url for 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Synchronizing submodule url for 'components/bt/controller/lib_esp32'
Synchronizing submodule url for 'components/bt/controller/lib_esp32c3_family'
Synchronizing submodule url for 'components/bt/host/nimble/nimble'
Synchronizing submodule url for 'components/cmock/CMock'
Synchronizing submodule url for 'components/esp_phy/lib'
Synchronizing submodule url for 'components/esp_wifi/lib'
Synchronizing submodule url for 'components/esptool_py/esptool'
Synchronizing submodule url for 'components/ieee802154/lib'
Synchronizing submodule url for 'components/json/cJSON'
Synchronizing submodule url for 'components/lwip/lwip'
Synchronizing submodule url for 'components/mbedtls/mbedtls'
Synchronizing submodule url for 'components/mqtt/esp-mqtt'
Synchronizing submodule url for 'components/openthread/lib'
Synchronizing submodule url for 'components/openthread/openthread'
Synchronizing submodule url for 'components/protobuf-c/protobuf-c'
Synchronizing submodule url for 'components/spiffs/spiffs'
Synchronizing submodule url for 'components/tinyusb/tinyusb'
Synchronizing submodule url for 'components/unity/unity'
Synchronizing submodule url for 'examples/build_system/cmake/import_lib/main/lib/tinyxml2'
Synchronizing submodule url for 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'
ncc@ncc-virtualbox:~/esp/esp-idf$ git submodule update --init
ncc@ncc-virtualbox:~/esp/esp-idf$ ./install.sh
Detecting the Python interpreter
Checking "python" ...
..
All done! You can now run:

  . ./export.sh

micropython/ports/esp32$ get_idf
micropython/ports/esp32$ idf.py build

make: Entering directory '/home/ncc/.virtualenvs/microPython/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK mpy-cross
   text    data     bss     dec     hex filename
 304451   17184     840  322475   4ebab mpy-cross
make: Leaving directory '/home/ncc/.virtualenvs/microPython/micropython/mpy-cross'
[963/1191] Generating ../../genhdr/qstr.i.last
FAILED: genhdr/qstr.i.last

robert-hh, 2022-03-03:
I miss in your log:

git checkout v4.4
git submodule update --init

You can issue these commands again. That does not hurt

nccwarp9, 2022-03-03:
changed formating in prior post. 
git checkout 4.4 was the first line

robert-hh, 2022-03-03:
git checkout v4.4

you missed the character v. 
v4.4 is an existing tag of a certain commit.

robert-hh, 2022-03-03:
And, by the way, what does the command get_idf? Is that setting the environment?

nccwarp9, 2022-03-03:
> And, by the way, what does the command get_idf? Is that setting the environment?

alias get_idf='. $HOME/esp/esp-idf/export.sh'

nccwarp9, 2022-03-03:
> git checkout v4.4
> 
> you missed the character v. v4.4 is an existing tag of a certain commit.

looks like it ... fraking one symbol, its always one . , v , compiling, will report

robert-hh, 2022-03-03:
After that get_idf (I have a different alias for it), I usually simply run:

make

nccwarp9, 2022-03-03:
solved!
compile success

Thanks !

closed 2022-03-03 by nccwarp9

TGiles1998, 2023-03-12:
@nccwarp9 Hi, I'm getting the same error. How did you solve this? Did you re-clone the esp-idf from master and then checkout the v4.4 branch? Many thanks,

Thomas

IvRogoz, 2023-03-18:
> 

check here if you find an answer. [https://warped3.substack.com/p/compiling-micropython-with-espressif](https://warped3.substack.com/p/compiling-micropython-with-espressif)
it was some time ago so I don't really remember how I fixed it

QGB, 2023-04-22:
idf.py -v build