micropython/micropython #8379: Main.py doesn’t load on initial power up

serbyxp, 2022-03-03:
My boot.py is in changed nothing in it. As per the documentation it says it doesn’t need to be changed. But on cold power up, ( just plugging it into USB port on computer) I get the repl, but none of my main.py loads. When I soft restart it ( pushing the button or ctrl d on thonny) then all the settings from main.py are there. Is there a way I can ensure that main.py loads by adding something in boot.py to call / load main.py?

i seen some other posts about it but I don’t think I’ve seen the right one. 
thanks

robert-hh, 2022-03-03:
There was a change sometime ago, that main.py is not run when executing boot.py failed. But that sequence is run both during hard boot and soft boot. Which board/firmware version do you use?

serbyxp, 2022-03-03:
I used the port from a repo from a guy called glen with esp32 and esp32 now I believe it is 1.17 on esp32 I have 3 diffrent esp32 boards that I am trying to make some sort of mesh one esp32 is basic esp32 4mb one the other is a ttgo T-display and the main one I’m using as the “hub or server” is ttgo t-pcie which has 16mb and 4mb we over E

robert-hh, 2022-03-03:
Ctr-B in REPL will tell you the firmware version. For ESP32, main.py will be called after boot.py even if there is an error in boot.py. But if there is an error in main.py, not all of the script will be executed. So you have to catch possible errors, e.g. with a try/except clause.

serbyxp, 2022-03-03:
It says v1.17-142-gbb9aac55a

serbyxp, 2022-03-03:
> 

Ok so in the boot.py I can add the try except clause, but that’s where I’m stuck . Like 

try 
(what?) try (wifi.isconnected()) or (main.py)

robert-hh, 2022-03-03:
in the esp32 port, main.py will always be executed if it exists. So it runs, and my guess is, that it fail early due to a hard error. About the try/except clause. that looks usually like:
```
try:
    do_something_that_may_fail_with_an_exception
except Exception as e:
   print("That something failed with error", e)
```
That would prevent  main.py to terminate early because of a hard error. "Soft" errors which just return a certain status code have to be handled by your code differently.
The __main__ in a python script does not apply in the ESP32 environment. That is something being relevant if code is called e.g. from a UNIX command line. But for that please refer to a Python text book.

robert-hh, 2022-03-03:
Question like your's are best discussed in the MicroPython forum. https://forum.micropython.org
The place is is the issue tracker for errors in the firmware. In your case it is more likely a usage topic.

serbyxp, 2022-03-03:
Thank you, I will look at the discussion board thank you.