micropython/micropython #8380: ESP32 esp32.Partition does not accept keywords

chrisovergaauw, 2022-03-03:
Hi,

it is documented that the ESP32 takes a keyword argument block_size: [here](https://docs.micropython.org/en/latest/library/esp32.html)

In practice I get:
```python
MicroPython v1.18-82-g20df071e3-dirty on 2022-03-03; CUSTOM with ESP32
import esp32
esp32.Partition(esp32.Partition.RUNNING, block_size=512)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function doesn't take keyword arguments
```

I need this functionality because the default 4096 is way too big for our application. Is there anything I can do to get this to work?

chrisovergaauw, 2022-03-04:
Hi again,

Re-opening because I just built and ran latest (665f0e2a68dfa4944f60e85a00d0543bb3d7fee8) and it still occurs.

chrisovergaauw, 2022-03-04:
I believe the issue is that it is not a keyword argument.

```python
>>> import esp32
>>> esp32.Partition(esp32.Partition.RUNNING, 512)
<Partition type=0, subtype=0, address=65536, size=2031616, label=factory, encrypted=0>

```

dpgeorge, 2022-03-04:
For simplicity, in the constructor `block_size` is positional only.  So it works if you just pass it in as positional.

chrisovergaauw, 2022-03-04:
Yea, I figured that out as well. Thank you.

dpgeorge, 2022-03-07:
Docs updated in bb2bd071f74c04237618958521e58d48d2dc35d4 to specify that the `Partition` constructor doesn't take keyword args.

closed 2022-03-07 by dpgeorge