micropython/micropython #8382: ESP32: feature request Partition

chrisovergaauw, 2022-03-04:
related to: [#8380](https://github.com/micropython/micropython/issues/8380)

it would be nice if Partition.get_next_update() would also accept the block_size argument.
This way you won't be locked in to a certain block_size after initial deployment when doing OTA updates.

dpgeorge, 2022-03-04:
`Partition.get_next_update()` should probably default to using the `block_size` of the parent `Partition` instance (not use `NATIVE_BLOCK_SIZE_BYTES` like it currently does).

> it would be nice if Partition.get_next_update() would also accept the block_size argument

I can't immediately see why that would be useful, to change the `block_size`, but it does make sense to generalise it to allow it to be passed as an optional argument.

chrisovergaauw, 2022-03-04:
One example which we are running into is the following:

With every new OTA update we’re doing, our application becomes more feature rich and as an effect more memory is consumed.
We’re now getting to a point where `Partition.get_next_update()` fails because it cannot reserve 4096 bytes.
Technically there’s enough RAM available, but due to fragmentation it’s not going to work. (This is an entirely different issue, but currently I view a lower block_size as a temporary/necessary workaround)
Memory fragmentation aside, just the growing feature set in and of itself could mean that in the future, you have less heap available and therefore are required to work with smaller a BLOCK_SIZE.