micropython/micropython #8383: Docu: Multiple machine.Timer() instances on ESP8266

manuel-91, 2022-03-04:
There is sadly only a [little docu on the timer usage on ESP8266](https://github.com/micropython/micropython/blob/master/docs/esp8266/quickref.rst#timers). It shows this example:
```python
from machine import Timer

tim = Timer(-1)
tim.init(period=5000, mode=Timer.ONE_SHOT, callback=lambda t:print(1))
tim.init(period=2000, mode=Timer.PERIODIC, callback=lambda t:print(2))
```

As it uses the same timer only "2" is printed every 2 seconds, the one_shot "1" is **never** printed. I tried two instances of `Timer(-1)` but still only the last one was active. I just tried `Timer(-2)` and it worked, with `Timer(1)` and `Timer(2)` it also worked:

```python
tim1 = Timer(1)
tim2 = Timer(2)
tim1.init(period=2500, mode=Timer.ONE_SHOT, callback=lambda t: print("once"))
tim2.init(period=1000, mode=Timer.PERIODIC, callback=lambda t: print("periodic"))
sleep(3.5)
tim2.deinit()

# expected printout:
# periodic
# periodic
# once
# periodic
```

 There is one thing I'm not sure about I'd like to understand before starting a pull request on the docu:
1. Is only one virtual (RTOS -based) timer possible on the ESP8266? [machine docu](https://docs.micropython.org/en/latest/library/machine.Timer.html#constructors) mentions Timer(-1) creates a virtual timer - which sounds good as it doesn't affect hardware.
2. If I create a timer with another ID, will it interfere with PWM or sleep() or time()?

By the way, great project, love it:)

eparadis, 2022-10-19:
I was curious about this same situation, and after looking at the C source code, the creation of the `Timer` object actually ignores it's parameter. It simply checks that you pass a single argument.

So what defines multiple `Timers`, I think, is simply creating multiple objects. All of them are created the same way, it seems.

eparadis, 2022-10-19:
If I'm not mistaken, the code in question is [here](https://github.com/micropython/micropython/blob/master/ports/esp8266/modmachine.c#L230).