micropython/micropython #8390: stm32/flashbdev: linker script flash storage layout should override the defaults in flashbdev.c

iabdalkader, 2022-03-07:
In 84969194a18ec015d2508db14be66457f123f687 flash storage layout configuration via linker script was introduced, but not all MCUs were converted to use it in 35e70c1698047170f9fb8b1edc65a7f7125f267f. The MCUs that still have defaults in `flashbdev.c`, for example `STM32F427xx`, can't use the linker script to configure the flash storage layout. Can all MCUs be switched to use the linker script approach ? Or alternatively, can a board-level config option be added, that when enabled gives precedence to the linker script layout ?

dpgeorge, 2022-03-08:
> Can all MCUs be switched to use the linker script approach ?

Yes.  This was the intention, I just didn't get around to it.

iabdalkader, 2022-03-08:
> Yes. This was the intention, I just didn't get around to it.

I can send a PR later to convert the remaining MCUs, following the others.

iabdalkader, 2022-03-13:
@dpgeorge Is this right ? It used to be the max flash sector, now it's the entire flash cache size:
```
#define FLASH_SECTOR_SIZE_MAX \
    (&_micropy_hw_internal_flash_storage_ram_cache_end[0] - &_micropy_hw_internal_flash_storage_ram_cache_start[0])
```

dpgeorge, 2022-03-16:
> Is this right ? It used to be the max flash sector, now it's the entire flash cache size:

Yes that looks correct, `FLASH_SECTOR_SIZE_MAX` should be the maximum amount of RAM available in the cache.

For the F746, it won't matter if this increases to 64k, it'll only ever use 32k because that's the maximum flash sector in the range used by the storage.

> Also this is wrong:

No, it's correct.  The comment says `16k(of 64k)` which is correct (only 16k of the 64k sector 4 is used).  128 = 16k*4 / 512.

iabdalkader, 2022-03-16:
> only 16k of the 64k sector 4 is used

But wouldn't this report the wrong FS size ?

dpgeorge, 2022-03-16:
The 128 is correct in the existing code.

But I see your point, that the new scheme that computes the number of blocks from the linker symbols won't work... I think what needs to be done is the `FLASH_FS` constants need to be decreased to reflect only the usable part of flash for the filesystem.

iabdalkader, 2022-03-16:
> the new scheme that computes the number of blocks from the linker symbols won't work.

Exactly.

iabdalkader, 2022-03-16:
> what needs to be done is the FLASH_FS constants need to be decreased to reflect only the usable part of flash for the filesystem.

Okay but the remainder of the sector can not be used for code, right ? The whole sector must be erased when updating the firmware.

dpgeorge, 2022-03-16:
> Okay but the remainder of the sector can not be used for code, right

Correct.

> The whole sector must be erased when updating the firmware.

Note that the filesystem is not touched when updating the firmware.  So it should be enough to just shrink `FLASH_FS` to the usable size.

iabdalkader, 2022-03-16:
> Note that the filesystem is not touched when updating the firmware

I know, I was just saying that the remainder of the sector will be unusable.

closed 2022-03-17 by iabdalkader