micropython/micropython #8392: ESP32, proximity sensor :D

Zibri, 2022-03-08:
I just installed micropython on my ESP32 WROOM32.
I was playing with it when I noticed this:
Run this program (press control E the paste the following code)
```
import esp32, machine	
try:
	print("\033[?25l")
	while(True):
		ad=machine.ADC(Pin(35)).read()
		if ad > 800:
			print("\033[F\033[AT = {0:8d}".format(ad))
		else:
			print("\033[F\033[A                ".format(ad))
except KeyboardInterrupt:
	print('End.\n\033[?25h')

```
now, the program will display nothing, normally, but as I move my hand near the main chip, the values of the ADC pins change!
the "noise" level is at about 0-600  so the program can be expanded to determine how close the hand is to the chip...
the value goes as high as 3000...

Any thoughts?

P.S.
I see the same behaviour on all ADC pins.

peterhinch, 2022-03-09:
This type of question should be raised in [the forum](http://[forum](http://forum.micropython.org/index.php).micropython.org/index.php) rather than in the MicroPython bug tracker. What you are observing is normal behaviour for a chip with CMOS inputs: open circuit inputs have very high impedance and are affected by capacitive coupling to charged bodies.

mattytrentini, 2023-10-24:
Closing, not a bug.

closed 2023-10-24 by mattytrentini