micropython/micropython #8393: Missing pin 20 on ESP32 definition

ladyada, 2022-03-08:
hiya for the esp32 pico v3 chip variant of the esp32, pin 20 is exposed:

[image]

but in the esp32 board definition this pin isnt available because WROVER and WROOM modules don't have this pin available
https://github.com/micropython/micropython/blob/master/ports/esp32/machine_pin.c#L96
shall i submit a PR to change this so the pin is valid?

dpgeorge, 2022-03-09:
> because WROVER and WROOM modules don't have this pin available

Is the reason they don't have it available because it's (1) just not wired through to the outside world; or (2) because it's used internally for something?

We had issues in the past where pins were enabled which were used for internal things like SPIRAM and so using them crashed the firmware.

> shall i submit a PR to change this so the pin is valid?

If using this pin on a WROVER/WROOM will not crash the firmware, then yes please!

ladyada, 2022-03-09:
looks like it isnt exposed on the QFN of original ESP32's...but what it was used for is unclear
https://esp32.com/viewtopic.php?t=6151
we could go with adding a new chip #define for the ESP32_PICO_V3_02 which has fixed flash, ram, different available pins...

dpgeorge, 2022-03-09:
> we could go with adding a new chip #define for the ESP32_PICO_V3_02

Would be good if there was already such a define in the IDF, but it doesn't look like there is.  Eg in components/soc/esp32/gpio_periph.c:
```c
    IO_MUX_GPIO19_REG,
    IO_MUX_GPIO20_REG, // This corresponding pin is only available on ESP32-PICO-V3 chip package
    IO_MUX_GPIO21_REG,
```
So I think just make it available on all ESP32 parts.  It should be safe to use it on a package that doesn't expose it, it just won't do anything.

ladyada, 2022-03-13:
yep! @kattni has kindly made a minimal PR :)
https://github.com/micropython/micropython/pull/8406

dpgeorge, 2022-03-21:
Fixed by 61c02e6500be7b174bb1c257f4798969fadab206

closed 2022-03-21 by dpgeorge

jerryneedell, 2022-03-23:
I tried the 20220323 nightly build for MP  on the feather esp32 v2 and I no longer get the invalid Pin error with scl = Pin(20)) but I am not able to access my I2C device 
```MPY: soft reboot
network config: ('10.0.0.161', '255.255.255.0', '10.0.0.1', '75.75.75.75')
MicroPython v1.18 on 2022-03-23; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
>>> 
>>> import pos
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pos.py", line 31, in <module>
  File "lsm6dsox.py", line 95, in __init__
  File "lsm6dsox.py", line 152, in __read_reg
OSError: [Errno 116] ETIMEDOUT
```  
 It works fine if I move back to scl on Pin(14) 

I tried a simple test -- connecting Pin 22 (SDA) to Pin 13 (which has a LED on it)  then 
```
from machine import Pin
p = Pin(22,Pin.OUT)   --- the RED LED comes on
p.value(False)        --- the RED LED goes off

then move to Pin 20 (SCL)
p = Pin(20,Pin.OUT)     --- no response on RED LED
p.value(False)          --- no response on RED LED
p.value(True)           --- no response on RED LED

also tried Pin 14  -- works as expected -- LED responds
```
  also reading back pin 20 after setting it to True or false always returns 0. Other pins return the expected value 
```
>>> p = Pin(20,Pin.OUT)
>>> p.value()
0
>>> p.value(True)
>>> p.value()
0
>>> p = Pin(22,Pin.OUT)
>>> p.value()
0
>>> p.value(True)
>>> p.value()
1
 ```