micropython/micropython #8402: Regex start of string replace seems to be repeatedly applied

Neradoc, 2022-03-09:
In MicroPython on ESP32 (v1.18 release and v1.18-202-gade2720e5).
Start-of-string `^a` regex in `sub` acts like it's applied repeatedly in some way.
```py
MicroPython v1.18 on 2022-03-09; ESP32 module with ESP32
Type "help()" for more information.
>>> import re
>>> re.sub('^ab', "0", "abababcabab")
'000cabab'
```
In C Python we get the expected result:
```py
>>> import re
>>> re.sub('^ab', "0", "abababcabab")
'0ababcabab'
```

dpgeorge, 2022-03-10:
Thanks for the report, I can confirm the issue.  The problem is that the `sub()` function keeps resetting (after a substitute) the start of the string that its scanning, so the `^` keeps matching.

A slightly more complicated example of the issue, showing that it's not simple to solve:
```python
>>> re.sub('(^ab|cab)', '0', 'abababcabab')
`00000`
```

dpgeorge, 2022-03-10:
Also similar issue for `split()`:
```python
print(re.compile('^ab').split('abababcabab'))
['', '', '', 'cabab']
```

dpgeorge, 2022-03-10:
I found a relatively simple fix; see #8403.

dpgeorge, 2022-03-16:
Fixed by 63f0e700f4e8927854ec1f70eeb114fa079632a3

closed 2022-03-16 by dpgeorge