micropython/micropython #8407: ESP32 : Setting password for AP_IF mode

AmirHmZz, 2022-03-10:
I'm trying to set password for `AP_IF` mode with these lines of code :

```Python
>>> import machine
>>> ap_wlan.active(True)
True
>>> ap_wlan.config(essid="EspAP", password="mytestpassword1234")
>>> ap_wlan.config("password")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: unknown config param
```

According to the source code, there should be a config param called password :
https://github.com/micropython/micropython/blob/da4b38e7562dfa451917f9d7f344a7f26de8c7bd/ports/esp32/network_wlan.c#L432-L440
But why I cannot use it? Am I doing it in a right way?

I should mention that I'm using [this release](https://micropython.org/resources/firmware/esp32-20220117-v1.18.bin) which is `micropython 1.18` on a `Wemos D1 R32` board which uses `ESP32-WROOM-32`.

vanroy-b, 2022-03-22:
This code works for me:
```# WIFI connection
ap = network.WLAN(network.AP_IF)
ap.active(True)
ap.config(essid='networkName', password='strongPassword', authmode=network.AUTH_WPA_WPA2_PSK)

while ap.active() == False:
    pass
print('Connection successful')
print(ap.ifconfig())```

notice the authmode argument, without it the network is unsecured for me
tested on micropython 1.17 with esp32-wroom-32 devkit v1

AmirHmZz, 2022-03-23:
Thank you @vanroy-b ! the only point for me was to config password before activating `AP_IF` mode. I'm going to close this issue.

closed 2022-03-23 by AmirHmZz