micropython/micropython #8408: Error with a script (RMT)

dnecode, 2022-03-12:
Hello,
I've started a project a few days ago, the aim is to control with a smartphone via bluetooth LE and an android app, that I created, a strip of leds. Everything works fine except that some leds are flashing with different colors randomly. I searched on the internet and I finally found it's a problem with RMT. Note that to control the leds, I need to use only this firmware esp32-idf3-20200411-v1.12-357.bin because the library I use [](https://www.gcworks.fr/tutoriel/esp/files/esp_ble_uart_v2.zip) works only with this firmware.

So I added this to my script :

`r = esp32.RMT(1, pin=machine.Pin(p), clock_div=80)
r
esp32.RMT(channel=1, pin=p, clock_div=80)
r.send_pulses((100, 2000, 100, 4000), start=0)`

but it leads me to an error :

`Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
OSError: ESP_ERR_INVALID_STATE
>
MicroPython v1.12 on 2019-12-20; ESP32 module with ESP32
Type "help()" for more information.
>>> `

Here is the full script :

`from esp_ble_uart import *
import machine, neopixel, time, esp32

n = 49
p = 13
np = neopixel.NeoPixel(machine.Pin(p),n)

r = esp32.RMT(1, pin=machine.Pin(p), clock_div=80)
r
esp32.RMT(channel=1, pin=p, clock_div=80)
r.send_pulses((100, 2000, 100, 4000), start=0)

nom = 'neonconnect'
UUID_UART = '6E400001-B5A3-F393-E0A9-E50E24DCCA9E'
UUID_TX = '6E400003-B5A3-F393-E0A9-E50E24DCCA9E'
UUID_RX = '6E400002-B5A3-F393-E0A9-E50E24DCCA9E'

uart = Bleuart(nom, UUID_UART, UUID_TX, UUID_RX)
uart.close()

val_rx = "0,0,0"

def rcp_rx():
    global val_rx
    val_rx = uart.read().decode().strip()
    print(val_rx)

uart.irq(handler=rcp_rx) # Interruption

while True :
    color = val_rx.split(",")
    print(color)
    r = int(color[0])
    g = int(color[1])
    b = int(color[2])
    for i in range(n):
        np[i] = (r, g, b)
    np.write()
    time.sleep(1)`

Please, I really need help.

Thx

mattytrentini, 2022-03-12:
There has been a _huge_ amount of work improving the NeoPixel control in recent releases of MicroPython. I know you've mentioned you need to use v1.12 but that was released in 2019; I can only suggest you update MicroPython and make changes to your library to support the newer release.

dnecode, 2022-03-13:
Thanks for the response. I'm still a newbie on micropython could you show me a tutorial, or explain to me how can I make changes to the library ?