micropython/micropython #8410: PWM on frequencies lower 611 Hz don't work on firmware 1.18

Kampfzwerg87, 2022-03-13:
I tried to using PWM on firmware 1.18 to control a servo motor and it doesn't work.
With high frequencies the PWM-Signal was ok, on lower frequencies I didn't get any signal.

So I did some test with an oscilloscope and realized that I don't get a PWM-Signal when I use frequencies lower than 611 Hz.
Downgrading to firmware 1.17 solved the problem.

I used the ESP32 NodeMCU Development Kit.

Here's the code I used to verify this issue:
```
import machine
import time

pwm = machine.PWM(machine.Pin(4), freq=1)

duty = 512
pwm.duty(duty)
PAUSE = 0.03
#Endlosschleife
while(True):
  for freq in range(1,1000):
    pwm.freq(freq)
    print(freq)
    time.sleep(PAUSE)
```
With firmware 1.17 it works fine, with firmware 1.18 it doesn't.

robert-hh, 2022-03-13:
Please try the most recent nightly build from the downloads web page. The problem should be fixed.

jej, 2022-03-13:
Solved here: https://github.com/micropython/micropython/commit/a5e64c209f2d76cb0cc903df875d87e8ca009507

leomlr, 2022-03-14:
Thanks for the fix.
I just tested and it still doesn't work with the downloadable version here: https://micropython.org/resources/firmware/esp32-20220117-v1.18.bin
Waiting the new version :)
Best regards

jej, 2022-03-14:
The fix was committed on February 1st... your firmware is dated January 17.
So try a nightly built here: https://micropython.org/download/esp32/
Hope it helps.

robert-hh, 2022-03-14:
Please use the recenet nightly build, as suggested: e.g. https://micropython.org/resources/firmware/esp32-20220314-unstable-v1.18-212-geec07332b.bin
Do not get confused by the 'unstable' in the file name. The nightly builds are pretty stable.

leomlr, 2022-03-15:
Thanks for responses. Now, it is working like a charm.

robert-hh, 2022-03-15:
Wonderful. Please do not forget to close the issue.

closed 2022-03-16 by dpgeorge