micropython/micropython #8412: Micropython won't boot on esp32 c3?

larsks, 2022-03-13:
I'm trying to use https://micropython.org/download/esp32c3-usb/ on a recently purchased ESP32 C3 module.  `flash_id` reports:

```
Serial port /dev/ttyUSB0
Connecting....
Chip is unknown ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 34:b4:72:4e:34:c4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

I'm writing the binary using the following command:

```
esptool.py --chip esp32c3 --port /dev/ttyUSB0 --baud 460800 \
  write_flash -z 0x0 esp32c3-usb-20220117-v1.18.bin
```

The write completes successfully, but the board doesn't boot into
micropython. Pressing the reset switch while connected to the serial
port results in:

```
Terminal ready
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6dc
load:0x403d0000,len:0x28c4
entry 0x403ce000
```

...and nothing else.

Both CircuitPython (https://downloads.circuitpython.org/bin/ai_thinker_esp32-c3s/en_US/adafruit-circuitpython-ai_thinker_esp32-c3s-en_US-7.2.0.bin) and the AT firmware install and boot successfully on this board.

dpgeorge, 2022-03-16:
You might want to try the non-USB board: https://micropython.org/download/esp32c3/ .  Be sure to try the latest unstable version as well.

Some C3 boards have native USB exposed, others have a USB-UART convertor.

larsks, 2022-03-23:
Thanks, that was it. I hadn't noticed there were two different builds available; sorry about that!

closed 2022-03-23 by larsks