micropython/micropython #8414: function doesn't take keyword arguments in mmicropython

SamRoy259, 2022-03-14:
>>> i=10
>>> i.to_bytes(1,byteorder='little') * 4

dpgeorge, 2022-03-16:
To save space, most of the built-in functions/methods do not take keyword arguments.

kalyan-2k19, 2023-04-30:
i2c.writeto(addr = 0x70, buf = 0x8050)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function doesn't take keyword arguments

I am also getting this error while working with i2c.

robert-hh, 2023-04-30:
According to the docs https://docs.micropython.org/en/latest/library/machine.I2C.html?highlight=i2c%20writeto#machine.I2C.writeto, the arguments are positional. And even if you drop the keywords, the second argument will cause an error, because it is not an object with  buffer protocol, but an int.