micropython/micropython #8415: [QUESTION] Compiling multiple USER_C_MODULES

Gruftgrabbler, 2022-03-14:
Hello. I tried to follow the compilation [instructions](https://docs.micropython.org/en/v1.14/develop/cmodules.html) in order to compile [uLAB](https://github.com/v923z/micropython-ulab) and a [sst7789 driver](https://github.com/russhughes/st7789_mpy)
together in one piece of MicroPython Firmware, targeting the rp2 pico. 

I extracted the following build file from the ulab repository and modified it for my needs. I am able to compile either ulab or st7789_mpy individually  with this file but not both at once.

Since I honestly don't really understand the short example in the documentation. I need to ask here for help.
What am I doing wrong here. What do I have to to to compile both modules at once?

```
#!/bin/bash

export BUILD_DIR=$(pwd)
export MODULES_DIR=$BUILD_DIR/modules

export ULAB_DIR=$MODULES_DIR/micropython-ulab/code
export ST7789_MPY_DIR=$MODULES_DIR/st7789_mpy

export MPY_DIR=$BUILD_DIR/micropython

if [ ! -d $MODULES_DIR ]; then
    mkdir modules
fi

# Clone modules into $MODULES_DIR
if [ ! -d $ULAB_DIR ]; then
    printf "Cloning ulab\n"
    ULAB_DIR=$BUILD_DIR/ulab/code
    cd $MODULES_DIR
    git clone https://github.com/v923z/micropython-ulab.git ulab
    cd ..
fi

if [ ! -d $ST7789_MPY_DIR ]; then
    printf "Cloning ST7789_MPY\n"
    cd $MODULES_DIR
    git clone https://github.com/russhughes/st7789_mpy st7789_mpy
    cd ..
fi

# Clone main micropython
if [ ! -d $MPY_DIR ]; then
    printf "Cloning MicroPython\n"
    git clone https://github.com/micropython/micropython.git micropython
fi

cd $MPY_DIR
git submodule update --init
cd ./mpy-cross && make # build cross-compiler (required)

cd $MPY_DIR/ports/rp2
rm -rf build-PICO
make submodule

export USER_CMODULES = $MODULES_DIR
export $CFLAGS_EXTRA="-DMODULE_ULAB_ENABLED=1 -DMODULE_ST7789_MPY_ENABLED=1"

make USER_C_MODULES=$USER_C_MODULES CFLAGS_EXTRA=$CFLAGS_EXTRA
```

Thank you and
regards
Gruftgrabbler

Gruftgrabbler, 2022-03-14:
Oh UPS I forgot to add a micropython.cmake file to the modules directory. Problem solved!

magicjohnson, 2022-12-28:
> Oh UPS I forgot to add a micropython.cmake file to the modules directory. Problem solved!

I have similar issue. Did you create a blank `micropython.cmake` ?