micropython/micropython #8418: REPL on Espressif ESP32-S2 board appears on wrong USB port

peterhinch, 2022-03-16:
I flashed the [generic ESP32-S2 firmware](https://micropython.org/resources/firmware/GENERIC_S2-20220117-v1.18.bin) to [this Espressif board](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-devkitm-1-v1.html). The board uses a UART-USB bridge to provide USB connectivity. Component placements to provide the internal USB interface are not populated.

Flashing was successful. On reset, the board's USB port emits debug information but no REPL. In [this forum thread](https://forum.micropython.org/viewtopic.php?f=18&t=12126) user ash10 explained that the REPL appears on the built-in software USB port on GPIO19-20. It seems that the board could be used by connecting a USB breakout to those pins (waiting for hardware to try this). This seems clunky.

I had naively assumed that the Espressif board would be a reference board as per ESP32. Is support for this board planned?

dpgeorge, 2022-03-17:
Since the S2 has native USB it was assumed that generic S2 boards would provide USB hardware, and so the GENERIC_S2 board was defined in this way, to have USB enabled on the REPL.

I don't have one of those Espressif S2 dev boards (because they were very hard to buy when they came out) and instead I use a UM FeatherS2 for development, which does have native USB wired up.

But it does make sense to support the Espressif S2 boards, even if they don't use native USB.  I can think of two options to do this:
1. enable UART and USB on the GENERIC_S2 board
2. have separate GENERIC_S2 and GENERIC_S2_USB board definitions

If it can be made to work, I'd rather option 1 because it's simpler.

@peterhinch, if possible could you please test the following patch on the S2 board you have:
```patch
--- a/ports/esp32/main.c
+++ b/ports/esp32/main.c
@@ -90,6 +90,7 @@ void mp_task(void *pvParameter) {
     #endif
     #if CONFIG_USB_ENABLED
     usb_init();
+    uart_stdout_init();
     #elif CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
     usb_serial_jtag_init();
     #else

```
This should enable UART REPL as well as USB REPL.

peterhinch, 2022-03-17:
@dpgeorge Unfortunately I've never succeeded in building ESP32 firmware.

jej, 2022-03-17:
Here: https://drop.infini.fr/r/zbXnqgnd14#ggkoTB/ZZBl1Vkw4sLo5u9NqYyd8RrmxIRitNwMO/I4=

```
idf.py -D MICROPY_BOARD=GENERIC_S2 -B build-GENERIC_S2  build
Executing action: all (aliases: build)
Running ninja in directory /usr/local/src/MicroPython/micropython/ports/esp32/build-GENERIC_S2
Executing "ninja all"...
[2/5] cd /usr/local/src/MicroPython/micropython/ports/esp32/build-G...icroPython/micropython/ports/esp32/build-GENERIC_S2/micropython.bin
micropython.bin binary size 0x11b6a0 bytes. Smallest app partition is 0x1f0000 bytes. 0xd4960 bytes (43%) free.
[3/5] Performing build step for 'bootloader'
[1/1] cd /usr/local/src/MicroPython/micropython/ports/esp32/build-GENERIC_S2/bootloader/esp-idf/esptool_py && /home/jej/.espressif/python_env/idf4.4_py3.9_env/bin/python /usr/local/src/MicroPython/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /usr/local/src/MicroPython/micropython/ports/esp32/build-GENERIC_S2/bootloader/bootloader.bin
Bootloader binary size 0x41f0 bytes. 0x2e10 bytes (41%) free.

Project build complete. To flash, run this command:
/home/jej/.espressif/python_env/idf4.4_py3.9_env/bin/python ../../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32s2  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x1000 build-GENERIC_S2/bootloader/bootloader.bin 0x8000 build-GENERIC_S2/partition_table/partition-table.bin 0x10000 build-GENERIC_S2/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    16880  (   11792 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1160864  (  870752 remaining)
total                  1226400
```

jej, 2022-03-17:
In fact both .bin are patched according to https://github.com/micropython/micropython/issues/8418#issuecomment-1069789380 suggestion. It's just that I did not know what .bin file from build dir was necessary (I always `deploy` with the command line).
Tell me if you need another file or compilation.

jej, 2022-03-17:
Note: I just compiled with the main.c modification. I didn't modify anything in the board definition. I don't know if `enable UART and USB on the GENERIC_S2 board` should be done before compilation.
Here is a firmware.bin from a clean compilation. It looks the same but in case...
https://drop.infini.fr/r/F3ZWNwWT75#4KCjiOmG3H0VHT96ZELMJ2TWGK8RBFT3jj74lFlD80E=

dpgeorge, 2022-04-14:
Fixed by fe9eaf12f391232f195ac052005ac2b852409dd6

closed 2022-04-14 by dpgeorge

peterhinch, 2022-04-16:
Now working well!

dpgeorge, 2022-04-18:
> Now working well!

Great!  Thanks for confirming.

matlawsky, 2022-08-26:
Hello, I encountered similar problem on a ESP32 - S2 - WROVER,  build v1.19.1 (2022-06-18). REPL is not showing up and I only get this:
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0xf5c
load:0x4004c000,len:0xaec
load:0x40050000,len:0x2be4
entry 0x4004c1b0

I'll try the generic one now but that should be brought to attention if it's easy to fix.