micropython/micropython #8420: Is possible to improve the RP2040 boot time?

beyonlo, 2022-03-16:
Hi all

Is possible to improve the boot time of RP2040 on MicroPython? I did tests here with scope and the time to turn on the led after power on the RPICO is the **178ms**.

**MicroPython version that I'm using on the RPICO.**
```
>>> import os
>>> os.uname()
(sysname='rp2', nodename='rp2', release='1.18.0', version='v1.18 on 2022-01-17 (GNU 11.2.0 MinSizeRel)', machine='Raspberry Pi Pico with RP2040')
>>> 
```
**This is my boot.py that I tested:**

```
from machine import Pin
p25 = Pin(25, Pin.OUT)
p25.on()
```

Any suggestion/idea?

Thank you very much.

dpgeorge, 2022-03-17:
The first thing to test would be to turn on the LED in C code at the start of `main()` (in main.c).  And then measure how long that takes.  That will give an idea whether the time spent is the chip starting up, or MicroPython starting up.

robert-hh, 2022-03-17:
131 ms from the rising edge of reset to a Pin toggle at the start of in main.c.

mattytrentini, 2022-03-24:
So @beyonlo, it seems like MicroPython can only make minimal boot time improvements here. 47ms to start the interpreter, read the filesystem, execute the code seems very reasonable. 

As I mentioned in Slack, you may be able to reduce some of that 47ms by freezing your code but there are diminishing returns at play since the bulk of the duration is in the micro physically starting... 

I'd suggest taking a look at the bootloader - are there any delays in UF2 that's slowing things down? - and possibly ask at the [Raspberry Pi Pico Forums](https://forums.raspberrypi.com/viewforum.php?f=143) to see if there are any ways to reduce the micro startup time. But in terms of MicroPython improvements I'm inclined to suggest they're unlikely to make a big dent. If you agree I'll close this ticket.

dpgeorge, 2022-03-24:
One thing you can try is modifying `sys.path` to put `".frozen"` first and remove `""` if it's not needed.  Multiple paths in `sys.path` will slow down some imports.

closed 2022-03-24 by beyonlo