micropython/micropython #8421: Micropython tests fail and then hang on Fedora x86_64

stratakis, 2022-03-17:
Hello,

On Fedora so far we have been [compiling](https://src.fedoraproject.org/rpms/micropython/blob/a647c9a094c9e3403791eef1221650893080120d/f/micropython.spec#_72) micropython in a way that utilizes the default compiler flags provided by the respective Makefile's, however when a [change](https://fedoraproject.org/wiki/Changes/SetBuildFlagsBuildCheck) was implemented distribution wide, to apply the default compiler flags everywhere, we stumbled upon some issues.

The compiler options added are:
```
-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection
```

More specifically, the first issue was the addition of -Wformat-security which required a missing -Wformat option. That was easily worked around by injecting the proper flag, however -Wformat could be a welcome addition to the various Makefile's.

After the workaround and compiling the interpreter, many tests failed and then the test run hanged on x86_64. This issue did not affect the i686 architecture.

I'm attaching a full build log of a successful and a failed run.

[build_log_default.txt](https://github.com/micropython/micropython/files/8278306/build_log_default.txt)
[build_log_flags.txt](https://github.com/micropython/micropython/files/8278426/build_log_flags.txt)

dpgeorge, 2022-03-17:
Thanks for the report.  Using the above flags I could reproduce the problem, and it looks like the issue is related to the NLR code (custom setjmp/longjmp handling).

As a workaround you can try the following:
```
make CFLAGS_EXTRA="-DMICROPY_NLR_SETJMP=1"
```

dpgeorge, 2022-03-17:
It looks like it's related to the `-flto=auto` option.  Simply adding this option to `CFLAGS` and compiling on x86-64 will lead to a crash in `nlr_push`.

If possible I'd suggest disabling LTO.  That would be better than the above suggestion of setting `MICROPY_NLR_SETJMP=1`.

stratakis, 2022-03-17:
Indeed disabling LTO makes the trick, thanks for figuring it out! It's also one of the flags that were recently [enabled by default](https://fedoraproject.org/wiki/LTOByDefault) for all the Fedora packages.

Would it be possible to make the NLR code handle properly the link time optimizations?

Nevertheless the workaround is sound.