micropython/micropython #8423: ESP32-S2 uasyncio console output stalls

peterhinch, 2022-03-17:
On occasion my application stops sending output to the console. Pressing Enter on the PC causes stalled output to appear. The application continues to run while the output is stalled.

Trivial test cases don't show the fault: it only seems to happen if there is a fair bit going on. This demo exhibits the fault.
```python
import uasyncio as asyncio

num_coros = (100, 200, 500, 1000)
iterations = [0, 0, 0, 0]
duration = 2  # Time to run for each number of coros
count = 0

async def foo():
    global count
    while True:
        await asyncio.sleep_ms(0)
        count += 1

async def test():
    global count
    old_n = 0
    for n, n_coros in enumerate(num_coros):
        print(f"Testing {n_coros} coros for {duration} secs")
        count = 0
        for _ in range(n_coros - old_n):
            asyncio.create_task(foo())
        old_n = n_coros
        await asyncio.sleep(duration)
        iterations[n] = count

async def report():
    await test()
    for x, n in enumerate(num_coros):
        print(f"Coros {n:4d}  Iterations/sec {round(iterations[x]/duration):5d}")

asyncio.run(report())
```
The problem occurs under mpremote, miniterm and rshell.

dpgeorge, 2022-03-17:
I cannot reproduce any issue with the above script.  I took a UM FeatherS2 and put the GENERIC_S2 firmware on it.  This uses the native USB port on the ESP32S2.

I'm at commit a16dcc8136c8af47d69e7a4b3a55270db0e4c637 with IDF v4.4.

I ran the above script in a loop via:
```
$ while true; do mpremote run test.py; done
```
It ran for 40 minutes without any problem.

peterhinch, 2022-03-18:
That one-liner stalls almost immediately here, also on a UM FeatherS2.

This is PC-related. The fault occurs on this 3 year old Dell Precision 5530 laptop which I use daily with various MP targets. On an ancient HP Zbook the test runs fine. This also explains the problems I had getting MP onto the UM FeatherS2. This gave me protocol errors despite using the latest `esptool.py`. I had to use the UF2 build. Evidently Dell's USB serial doesn't quite match Espressif's (although ESP32 and ESP8266 are fine).

Incidentally I was underwhelmed by the fact that UM's website directs you to an IRC channel for support.

peterhinch, 2022-03-18:
Forget my script - the quick way to demo this fault is to erase flash. On a "good" PC it works. On a "bad" one `esptool.py` issues a protocol failure.

The workround is to use another PC but it had me tearing my hair out for some time yesterday. I'll post a warning in the forum.

Is this an Espressif issue? What is the way forward?

mattytrentini, 2022-04-18:
Can't comment on the actual issue but knowing Seon (UM) I was surprised with:

>Incidentally I was underwhelmed by the fact that UM's website directs you to an IRC channel for support.

That's not actually IRC - that's a reference to a channel in _UM's Discord server_. He spends a lot of time supporting folks trying to use MicroPython on his devices! He will direct them to the forum if he can't help them.