micropython/micropython #8430: ESP8266 socket.recv(1024) doesn't get all data

xboxplayer9889, 2022-03-19:
ESP8266  (ESP8266EX, ESP-12F, [v1.18 (2022-01-17) .bin](https://micropython.org/resources/firmware/esp8266-20220117-v1.18.bin)) 

socket.recv() doesn't get all data on POST

with GET I can receive all the data, but with POST if the request size bigger than 536byte I receive the first 536, and no more come, repeated recv hangs/waits too much, no error, but page cannot be served.

xboxplayer9889, 2022-03-20:
Although socket.recv() don't works well for me,
I have inspected this example and it helped me to rewrite my code to get any info (all the HEAD data with cookies, POST): 
https://github.com/micropython/micropython/blob/master/examples/network/http_server.py

dpgeorge, 2022-03-20:
This is expected behaviour, `socket.recv(n)` reads at most `n` bytes from the stream, but maybe less.  It will not block if there is at least one 1 byte to return.

I see you have found a way forward, so will close the issue.

closed 2022-03-20 by dpgeorge

xboxplayer9889, 2022-03-21:
Thanks for answering,

- first of all,  I experienced it only with POST
- after my tries I can say recv() blocks every time, maybe only if set_nonblocking enabled, but that needs different code to handle not ended receive.
- I tried to receive 32 bytes at once but the last 32 byte always hangs even 1-31 received.
- And I tried at Python3.8 on my Linux and that always receive all the data if the buffer size is long enough.

But I'm happy I have learnt more about HTTP packets and I think not recv(bigbuffer) is the good solution....
Ty