micropython/micropython #8431: rp2: No way to avoid conflicting claims on DMA channels

nickovs, 2022-03-20:
At the moment there is no way on the RP2 port to write properly modular code that makes use of DMA without running the risk of two modules both trying to use the same channel. The underlying Raspberry Pi Pico C/C++ SDK provides a library function `dma_claim_unused_channel()` to deal with exactly this problem, but it is not exposed in the MicroPython. The lack of support for this is an impediment to writing anything other than trivial code that uses DMA.

The `rp2.DMA` class implemented in #7641 fixes this issue and at the same time provides a high level, Pythonic interface to the DMA hardware. It would be **really** helpful if this PR could be merged.

mdaeron, 2023-03-04:
Has there been any recent change re: "using DMA without running the risk of two modules both trying to use the same channel"? In practice, what are my options today to avoid this problem?

robert-hh, 2023-03-04:
At C level there is a set of functions to claim unused DMA channels, checking, whether channels are claimed etc.
There are two PRs to create DMA support classes #7641 (Aug 2021) and #10704. (Feb 2022). #7641 would support what you need. But it was never merged, which seems to be a pity.

nickovs, 2023-03-04:
Sorry, I sort of let #7641 drop due to lack of time. @dpgeorge asked me to change the way that the configuration is passed in and I've simply not had the chance to do that refactoring. It's been on my mind recently and I hope to get to it soon. In the mean time if you are willing to recompile the code yourself the patch works just fine as it stand, it just has a slightly clunky system for passing in the configuration.