micropython/micropython #8432: AttributeError: 'function' object has no attribute '__name__'

warnerjon12, 2022-03-20:
Some (maybe all) of the builtin functions do not have a `__name__` attribute like regular functions:

```
>>> def f(x):
...     return x
... 
>>> f.__name__
'f'
>>> from math import sin
>>> sin.__name__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'function' object has no attribute '__name__'
```

Additionally, micropython does not currently allow assignment/reassignment of function attributes like CPython does:

**micropython**
```
>>> f.__name__ = 'g'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'function' object has no attribute '__name__'
>>> f.some_attr = 'some value'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'function' object has no attribute 'some_attr'
>>> sin.__name__ = 'g'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'function' object has no attribute '__name__'
>>> sin.some_attr = 'some value'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'function' object has no attribute 'some_attr'
```
**CPython**
```
>>> f.__name__ = 'g'
>>> f.some_attr = 'some value'
>>> sin.__name__ = 'g'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: attribute '__name__' of 'builtin_function_or_method' objects is not writable
>>> sin.some_attr = 'some value'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'builtin_function_or_method' object has no attribute 'some_attr'
```

stinos, 2022-03-20:
Some of this is documented: http://docs.micropython.org/en/latest/genrst/core_language.html#user-defined-attributes-for-functions-are-not-supported. Note that some ports also don't implement `__name__` at all see e.g. https://github.com/micropython/micropython/issues/2594.

dpgeorge, 2022-06-24:
See also #601 (the `__doc__` attribute is not implemented).

This is just a limitation of MicroPython, it is designed to be small and doesn't implement everything that CPython does.  In this case it's because it would require lots more RAM to have custom attributes, and lots more ROM to have the name of all functions included in the firmware.

Unless there's some really compelling reason to implement these things, they will remain unimplemented.

closed 2022-06-24 by dpgeorge