micropython/micropython #8436: mpy_ld/esp32: cannot handle reloc-type #59 (R_XTENSA_PDIFF32) on idf 4.3.1

mgliewe, 2022-03-22:
Hi,
while playing around, i found that on esp32 the native examples wouldn't build.

mpy_ld complaint about an unknow relocation type 59 (denoted R_XTENSA_PDIFF32 in xtensa bfd sources).

After some digging around in the xtensa bfd sources, i found out, these seems to be an unsigned version of the already used relocation type  R_XTENSA_DIFF32, so just dealing equaly seemed to fix that problem. Unfortunately I didn't realy found any docs about these relocation entries, so thats more or less a guess.

lg maG

```
$ idf.py --version
ESP-IDF v4.3.1-dirty
$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
```

dpgeorge, 2022-06-09:
Thanks for the report and investigation.  The native examples build fine with IDF v4.2.x which uses an older toolchain.

I think you are right that this new type 59 can be treated the same as the existing R_XTENSA_DIFF32.  See eg http://lists.busybox.net/pipermail/buildroot/2020-May/282853.html

dpgeorge, 2022-06-09:
Fixed by abb38503987f2523726d05819e87977e3cdd65b5.  CI added in 313f082896bbf969dd91497151419efa6c42c148

closed 2022-06-09 by dpgeorge