micropython/micropython #8440: Unable to use pin 20 as SCL for I2C on ESP32.

kattni, 2022-03-23:
I submitted the fix to #8393 to expose pin 20 on ESP32. On the Adafruit Feather ESP32 V2, SCL is pin 20. I loaded the latest nightly build onto my board. I am running into the following issues.

When trying to initalise I2C, it is not accepting pin 20 as a viable option for SCL. 
```
>>> from machine import Pin, I2C
>>> i2c = I2C(0, scl=Pin(20), sda=Pin(22))
E (2540280) i2c: /home/micropython/esp-idf-v4.2/components/driver/i2c.c:767 (i2c_set_pin):scl gpio number error
```
I2C set up works with pin 14 as SCL.

If I try to simply init pin 20 as an output, it also fails.
```
>>> from machine import Pin
>>> p20 = Pin(20, Pin.OUT)
E (19500) gpio: gpio_set_direction(263): GPIO number error
```
@jepler suggested that perhaps the issue is resolved in the ESP IDF v4.4, but an attempt to build MicroPython with that version resulted in a failure to fit in flash. So, I have not tested this possibility.

kattni, 2022-03-23:
This is evidently definitely related to the EDP IDF. MicroPython would need to be built with v4.4 to get pin 20 to work properly. I will make a note in our documentation that as-is, pin 20 won't work with MicroPython.

If you have any plans for updating the ESP IDF, please let me know here. Regardless, feel free to close this issue if it is not worth continuing to track.

dpgeorge, 2022-03-23:
We do need to fix/improve the build with IDF 4.4 so all boards can use this latest IDF version.  The main thing is to support multiple heaps in the GC, #3580.

jerryneedell, 2022-03-24:
FYI -- with my feather esp32 V2 I am experiencing a different issue: I can configure I2C with scl on Pin 20:

I have loaded  esp32spiram-20220323-unstable-v1.18-244-g66fe3d5cb.bin

```
MicroPython v1.18 on 2022-03-23; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
>>> from machine import Pin,I2C
>>> i2c = I2C(0, scl = Pin(20), sda = Pin(22))
>>> 
```

but  then any attempt to utilize the i2c bus fails:
```
MPY: soft reboot
network config: ('10.0.0.161', '255.255.255.0', '10.0.0.1', '75.75.75.75')
MicroPython v1.18 on 2022-03-23; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
>>> from machine import Pin, I2C
>>> i2c = I2C(0,scl=Pin(20),sda=Pin(22))
>>> from lsm6dsox import LSM6DSOX
>>> 
>>> lsm = LSM6DSOX(i2c, gyro_odr=26, accel_odr=26, gyro_scale=250, accel_scale=2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lsm6dsox.py", line 95, in __init__
  File "lsm6dsox.py", line 152, in __read_reg
OSError: [Errno 116] ETIMEDOUT
>>> 
```
an i2c.scan() just hangs:
```
MPY: soft reboot
network config: ('10.0.0.161', '255.255.255.0', '10.0.0.1', '75.75.75.75')
MicroPython v1.18 on 2022-03-23; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
>>> from machine import Pin, I2C
>>> i2c = I2C(0,scl=Pin(20),sda=Pin(22))
>>> i2c.scan()

``
I had to power cycle  the board to continue. 

with nothing connected to Pin 20, I am not able to reliably control the output state
```
 MPY: soft reboot
network config: ('10.0.0.161', '255.255.255.0', '10.0.0.1', '75.75.75.75')
MicroPython v1.18 on 2022-03-23; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
>>> from machine import Pin
>>> p20 = Pin(20,Pin.OUT)
>>> p20.value()
0
>>> sum(p20.value() for _ in range(1000))
147
>>> p20.value(1)
>>> sum(p20.value() for _ in range(1000))
111
>>> sum(p20.value() for _ in range(1000))
131
>>> p20.value(0)
>>> sum(p20.value() for _ in range(1000))
134
>>> 
```

I don't know why my results differ from those reported bu @kattni 
This is the same board.

mopore, 2022-08-07:
I also get an "OSError: [Errno 116] ETIMEDOUT" with a "feather esp32 V2" and MicroPython v1.19.1
Using pin 14 walks but is not the solution since I want to use the stacked FeatherWing OLED Display.