micropython/micropython #8444: ESP8266 - UART drops 16th digit or randomly stops receiving after first 16 digits

Trikkitt, 2022-03-24:
On 1.18 (and earlier versions, unsure which one introduced this bug) the UART receive loses data.  Even when settings the RxBuf value to a 16 or a high number, the cost randomly loses the 16th digit it receives or simply ends the receive at 16 digits despite there being more digits arriving.

jonnor, 2024-09-08:
Does this problem still exist on current MicroPython, version 1.23 or later?

robert-hh, 2024-09-08:
The relation between the buffer size set with `rxbuf=n` and the number of bytes it can hold is a little bit inconsistent between ports. The ringbuf implementation used by most ports can hold n-1 bytes if the size is set to n.  A port may cater for that in the UART.init() method. That said, with `rxbuf=n` each port can hold at least n-1 bytes in the receive buffer. 
The default for ESP8266 is 16, and it can hold 15 bytes.