micropython/micropython #8445: missing file : ports/unix/modos.c

AhmedHosamZaghloul, 2022-03-24:
Hi,
using  Cygwin on windows,
when i try to run make on the embedding example i get this error
```
make[1]: *** No rule to make target 'ports/unix/modos.c', needed by 'build/genhdr/qstr.i.last'.  Stop.`
```
there is a file with a similar name "moduos.c", i tried renaming it and running again , i got this error
```
../../ports/unix/main.c: In function ‘main_’:
../../ports/unix/main.c:500:16: error: ‘MICROPY_PY_SYS_PATH_DEFAULT’ undeclared (first use in this function); did you mean ‘MICROPY_PY_SYS_PATH_ARGV_DEFAULTS’?
  500 |         path = MICROPY_PY_SYS_PATH_DEFAULT;
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                MICROPY_PY_SYS_PATH_ARGV_DEFAULTS
../../ports/unix/main.c:500:16: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [../../py/mkrules.mk:77: build/ports/unix/main.o] Error 1
```

stinos, 2022-03-24:
> using Cygwin on windows,

Following the instructions https://github.com/micropython/micropython/tree/master/ports/windows#building-under-cygwin on a clean Windows 10 installation builds without problems. Can you post the differences for your setup?

dpgeorge, 2022-03-24:
It's a problem with the *embedding example*.  I really thought we had some CI that built this... but apparently not!

AhmedHosamZaghloul, 2022-03-24:
thanks for the quick reply,
will this problem be fixed soon?

dpgeorge, 2022-03-25:
Fixed by b083cdba2b59e7ad68a718cec9d015ab7aede4c8

closed 2022-03-25 by dpgeorge