micropython/micropython #8446: ESP32 - Buffered serial bytes

Tangerino, 2022-03-24:
### Use case
Send and receive serial data as fas as possible on a loopback serial port (tx <-> rx)
Target device: ESP32
uPython V1.18

### Test code
```python
from time import ticks_diff
from time import ticks_ms
import machine

# Serial port TX pin 12 is connected to the RX pin 34 of the ESP32

def test(baud_rate):
    uart = machine.UART(1, tx=12, rx=34, timeout=10, baudrate=baud_rate)
    now = ticks_ms()
    uart.write(b"U")
    while True:
        if uart.any():
            rx = uart.read()
            elapsed = ticks_diff(ticks_ms(), now)
            print(f"Got {len(rx)} bytes - Elapsed {elapsed} - {rx}")
            break

for baud_rate in [110, 300, 9600, 19200, 38400]:
    print(f"Testing baud rate {baud_rate}")
    test(baud_rate)
```

### Expected behavior
Receive back, as soon as possible, the data sent.
On a 110bps, 10 bits per byte the expected time is 1 / 110 * 10 => 90 msec
WIth 9600 bps => 1 msec

### Results
```
Testing baud rate 110
Got 1 bytes - Elapsed 1036 - b'U'
Testing baud rate 300
Got 1 bytes - Elapsed 386 - b'U'
Testing baud rate 9600
Got 1 bytes - Elapsed 26 - b'U'
Testing baud rate 19200
Got 1 bytes - Elapsed 26 - b'U'
Testing baud rate 38400
Got 1 bytes - Elapsed 16 - b'U'
```

### Observations
The data takes much longer to get back to the main program than expected, specially in lower baud rates.

robert-hh, 2022-03-25:
Making a few test it appears, that the  time is spent for any() returning True. After a call to write, data is sent immediately, and after any reports data presence, read returns data immediately. uart.any() just calls uart_get_buffered_data_len(), so it's somewhere in the esp IDF slowing it down.
P.S.: Changing the code using uart.read() instead of uart.any() does not change the timing.

peterhinch, 2022-03-25:
Indeed. See [this example](https://github.com/peterhinch/micropython-async/blob/master/v3/as_demos/auart.py) for a way to do concurrent send and receive.

Tangerino, 2022-03-25:
Very intriguing @peterhinch 
Your code works for `hello\n` but not for `hello` (not a single byte received)
I'm using a byte oriented protocol, like modbus and not a string oriented protocol.
So the `\n` may or may not be present in the stream.
Any idea?

[image]

peterhinch, 2022-03-25:
My example is for newline-terminated strings.

There are two main options with serial data streams: the data may or may not have a delimiter (a character that terminates a string and is guaranteed not to appear otherwise). With a delimiter the receiver inputs data until the delimiter appears and returns that string. If there is no delimiter then the receiver must know in advance how many characters to expect and read that precise number.

Some byte-oriented protocols transmit the number of bytes in a fixed length field, followed by the bytes themselves. Note that I have no knowledge of the modbus protocol - for detailed help on that I suggest you raise a query in [the forum](http://[forum](http://forum.micropython.org/index.php).micropython.org/index.php).

Tangerino, 2022-03-25:
You are right about the 'know in advance' thing.
But I'm using devices with more than 30 of duty and they don't have a 'protocol' they are more like Morse Code.
IMH this is bad because there are some old protocols that use delimiters as a set of characters such as STX DLE, DLE DLE, DLE ETX.

Note the problem in my initial statement it works somehow but it takes one second to read 8 bytes with low baud rates. Which may impact the timeout handling, even if you know how many bytes should arrive.

I'll investigate ESP UART deeper.

About the forum, I even sent a message to the admin and also to @dpgeorge asking why I can't post anything but no reply I got :(

Thank you for the help.

closed 2022-03-25 by Tangerino

robert-hh, 2022-03-25:
You should be able to post on the forum. Only the first two posts have to be manually approved, which may take a few moments, but that should have happened by now.

Tangerino, 2022-03-26:
@robert-hh 
I fell bad, I don't find the new topic option
[image]

robert-hh, 2022-03-26:
You have to enter one of the sections, like ESP32, and then there is a new topic button.