micropython/micropython #8447: Compilation issue depend on the board type - ESP32

Tangerino, 2022-03-24:
Trying to compile on

- MacOS
- IDF v4.4
- uPython: v1.18 (latest)
- Port: ESP32

---
Compiling using `make` works
```
. . .
MPY websocket_helper.py
MPY neopixel.py
GEN /Users/projects/micropython/ports/esp32/build-GENERIC/frozen_content.c
[1351/1352] Generating binary image from built executable
esptool.py v3.2-dev
Merged 25 ELF sections
Generated /Users/projects/micropython/ports/esp32/build-GENERIC/micropython.bin
[1352/1352] cd /Users/projects/micropython/ports/esp32/build-GENERIC/esp-...bin /Users/projects/micropython/ports/esp32/build-GENERIC/micropython.bin
micropython.bin binary size 0x166040 bytes. Smallest app partition is 0x1f0000 bytes. 0x89fc0 bytes (28%) free.

Project build complete. To flash, run this command:
/Users/projects/.espressif/python_env/idf4.4_py3.9_env/bin/python ../../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build-GENERIC/bootloader/bootloader.bin 0x8000 build-GENERIC/partition_table/partition-table.bin 0x10000 build-GENERIC/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    21584  (    7088 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1466432  (  565184 remaining)
total                  1531968
```

---
Compiling using `make BOARD=GENERIC_SPIRAM`
```
❯ make BOARD=GENERIC_SPIRAM
idf.py -D MICROPY_BOARD=GENERIC_SPIRAM -B build-GENERIC_SPIRAM  build
Executing action: all (aliases: build)
Running ninja in directory /Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM
Executing "ninja all"...
[1/211] cd /Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM/esp-idf/main && echo -n
-n
[3/11] Performing build step for 'bootloader'
[1/1] cd /Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM/bootloader/esp-idf/esptool_py && /Users/projects/.espressif/python_env/idf4.4_py3.9_env/bin/python /Users/projects/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM/bootloader/bootloader.bin
Bootloader binary size 0x5450 bytes. 0x1bb0 bytes (25%) free.
[4/9] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/__/frozen_content.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/__/__/frozen_content.c.obj
/Users/projects/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DFFCONF_H=\"/Users/projects/micropython/lib/oofatfs/ffconf.h\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_ESP_IDF_4=1 -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -D__DBINTERFACE_PRIVATE=1 -Dvirt_fd_t="void*" -I/Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM/config -I/Users/projects/micropython -I/Users/projects/micropython/lib/berkeley-db-1.xx/PORT/include -I/Users/projects/micropython/ports/esp32 -I/Users/projects/micropython/ports/esp32/boards/GENERIC_SPIRAM -I/Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM -I/Users/projects/esp-idf/components/newlib/platform_include -I/Users/projects/esp-idf/components/freertos/include -I/Users/projects/esp-idf/components/freertos/include/esp_additions/freertos -I/Users/projects/esp-idf/components/freertos/port/xtensa/include -I/Users/projects/esp-idf/components/freertos/include/esp_additions -I/Users/projects/esp-idf/components/esp_hw_support/include -I/Users/projects/esp-idf/components/esp_hw_support/include/soc -I/Users/projects/esp-idf/components/esp_hw_support/include/soc/esp32 -I/Users/projects/esp-idf/components/esp_hw_support/port/esp32/. -I/Users/projects/esp-idf/components/heap/include -I/Users/projects/esp-idf/components/log/include -I/Users/projects/esp-idf/components/lwip/include/apps -I/Users/projects/esp-idf/components/lwip/include/apps/sntp -I/Users/projects/esp-idf/components/lwip/lwip/src/include -I/Users/projects/esp-idf/components/lwip/port/esp32/include -I/Users/projects/esp-idf/components/lwip/port/esp32/include/arch -I/Users/projects/esp-idf/components/soc/include -I/Users/projects/esp-idf/components/soc/esp32/. -I/Users/projects/esp-idf/components/soc/esp32/include -I/Users/projects/esp-idf/components/hal/esp32/include -I/Users/projects/esp-idf/components/hal/include -I/Users/projects/esp-idf/components/hal/platform_port/include -I/Users/projects/esp-idf/components/esp_rom/include -I/Users/projects/esp-idf/components/esp_rom/include/esp32 -I/Users/projects/esp-idf/components/esp_rom/esp32 -I/Users/projects/esp-idf/components/esp_common/include -I/Users/projects/esp-idf/components/esp_system/include -I/Users/projects/esp-idf/components/esp_system/port/soc -I/Users/projects/esp-idf/components/esp_system/port/public_compat -I/Users/projects/esp-idf/components/esp32/include -I/Users/projects/esp-idf/components/xtensa/include -I/Users/projects/esp-idf/components/xtensa/esp32/include -I/Users/projects/esp-idf/components/driver/include -I/Users/projects/esp-idf/components/driver/esp32/include -I/Users/projects/esp-idf/components/esp_pm/include -I/Users/projects/esp-idf/components/esp_ringbuf/include -I/Users/projects/esp-idf/components/efuse/include -I/Users/projects/esp-idf/components/efuse/esp32/include -I/Users/projects/esp-idf/components/vfs/include -I/Users/projects/esp-idf/components/esp_wifi/include -I/Users/projects/esp-idf/components/esp_event/include -I/Users/projects/esp-idf/components/esp_netif/include -I/Users/projects/esp-idf/components/esp_eth/include -I/Users/projects/esp-idf/components/tcpip_adapter/include -I/Users/projects/esp-idf/components/esp_phy/include -I/Users/projects/esp-idf/components/esp_phy/esp32/include -I/Users/projects/esp-idf/components/esp_ipc/include -I/Users/projects/esp-idf/components/app_trace/include -I/Users/projects/esp-idf/components/esp_timer/include -I/Users/projects/esp-idf/components/app_update/include -I/Users/projects/esp-idf/components/spi_flash/include -I/Users/projects/esp-idf/components/bootloader_support/include -I/Users/projects/esp-idf/components/bt/common/osi/include -I/Users/projects/esp-idf/components/bt/include/esp32/include -I/Users/projects/esp-idf/components/bt/common/api/include/api -I/Users/projects/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/Users/projects/esp-idf/components/bt/common/btc/profile/esp/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include -I/Users/projects/esp-idf/components/bt/host/nimble/port/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/include -I/Users/projects/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include -I/Users/projects/esp-idf/components/bt/host/nimble/esp-hci/include -I/Users/projects/esp-idf/components/nvs_flash/include -I/Users/projects/esp-idf/components/mbedtls/port/include -I/Users/projects/esp-idf/components/mbedtls/mbedtls/include -I/Users/projects/esp-idf/components/mbedtls/esp_crt_bundle/include -I/Users/projects/esp-idf/components/esp_adc_cal/include -I/Users/projects/esp-idf/components/mdns [... cut ...]
                 from /Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM/frozen_content.c:16:
/Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM/genhdr/qstrdefs.generated.h:987:6: note: previous definition of 'MP_QSTR_unlink' was here
 QDEF(MP_QSTR_unlink, 36862, 6, "unlink")
      ^~~~~~~~~~~~~~
/Users/projects/micropython/py/qstr.h:41:34: note: in definition of macro 'QDEF'
 #define QDEF(id, hash, len, str) id,
                                  ^~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [all] Error 2
```

robert-hh, 2022-03-24:
Try to remove the previous builds before compiling again.

Tangerino, 2022-03-24:
Errors persists after cleaning

Tangerino, 2022-03-24:
Unix port also works.

dpgeorge, 2022-03-25:
Try building `mpy-cross` explicitly before building the esp32 firmware (`make -C mpy-cross` from the root of this repo).

Tangerino, 2022-03-25:
Same error
---
```
❯ make -C mpy-cross clean
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -f mpy-cross
rm -f mpy-cross.map
rm -rf build
```
---
```
❯ make -C mpy-cross
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
. . .
. . .
. . .
CC gccollect.c
CC ../shared/runtime/gchelper_generic.c
LINK mpy-cross
__TEXT	__DATA	__OBJC	others	dec	hex
311296	16384	0	4295016448	4295344128	10005c000
```
---
```
❯ cd ports/esp32
❯ make BOARD=GENERIC_SPIRAM
idf.py -D MICROPY_BOARD=GENERIC_SPIRAM -B build-GENERIC_SPIRAM  build
Executing action: all (aliases: build)
. . .
. . .
. . .
/Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM/frozen_content.c:118:5: error: redeclaration of enumerator 'MP_QSTR_unlink'
     MP_QSTR_unlink,
     ^~~~~~~~~~~~~~
In file included from /Users/projects/micropython/py/obj.h:33,
                 from /Users/projects/micropython/py/objint.h:30,
                 from /Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM/frozen_content.c:16:
/Users/projects/micropython/ports/esp32/build-GENERIC_SPIRAM/genhdr/qstrdefs.generated.h:987:6: note: previous definition of 'MP_QSTR_unlink' was here
 QDEF(MP_QSTR_unlink, 36862, 6, "unlink")
      ^~~~~~~~~~~~~~
/Users/projects/micropython/py/qstr.h:41:34: note: in definition of macro 'QDEF'
 #define QDEF(id, hash, len, str) id,
                                  ^~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [all] Error 2
```

Tangerino, 2022-03-25:
Additional info: **MacMini M1**

Tangerino, 2022-03-30:
Anyone?

Tangerino, 2022-03-31:
Thank you.

The above command throws a new set of error messages.

But using `idf.py  build` works fine

closed 2022-03-31 by Tangerino