micropython/micropython #8453: ESP32S2 OR

Wind-stormger, 2022-03-25:


dpgeorge, 2022-03-28:
The block size was changed for a reason, and tested.

Can you explain exactly the steps you took to program your board, and when the error occurred?  Explain so that someone else can try to reproduce the same error.

FuHu-QC, 2022-03-29:
It seems that ffat partition (assigned with block_size = 512) can not be formated with littleFS2 .

``` python
>>>
MPY: soft reboot
MicroPython v1.18 on 2022-03-29; ESP32S3 module (spiram) with ESP32S3
Type "help()" for more information.
>>> import uos
>>> from esp32 import Partition
>>> uos.umount("/")
>>> bdev = Partition.find(Partition.TYPE_DATA, label="ffat", block_size=512)
>>> bdev
[<Partition type=1, subtype=129, address=4521984, size=3866624, label=ffat, encrypted=0>]
>>> bdev = bdev[0]
>>> uos.VfsLfs2.mkfs(bdev)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] EINVAL
>>>
```

For a fresh Micropython UF2 firmware installation, the _boot.py will be stopped  while formating ffat partition at https://github.com/micropython/micropython/blob/master/ports/esp32/modules/inisetup.py#L36

k-pa, 2023-01-04:
I am facing similar issue using 1.19.1 on Adafruit ESP32-S3 Feather with TinyUF2 Bootloader 0.9.2-2-g773c05a
After flashing a brand new board using uf2 firmware I can't copy files using mpremote or write file using REPL:
```
>>> f = open("test", "w")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```
I can see the following partitions which seems correct:
```
print(Partition.find(Partition.TYPE_DATA))
[<Partition type=1, subtype=2, address=36864, size=20480, label=nvs, encrypted=0>, 
<Partition type=1, subtype=0, address=57344, size=8192, label=otadata, encrypted=0>, 
<Partition type=1, subtype=129, address=3211264, size=983040, label=ffat, encrypted=0>]
```

Then :
```
inisetup.setup()
Performing initial setup
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "inisetup.py", line 36, in setup
OSError: [Errno 22] EINVAL
```

So I tried as suggested above :
```
>>> os.VfsFat.mkfs(bdev)
>>> vfs = os.VfsFat(bdev)
>>> os.mount(vfs, "/")

f = open("test.txt", "w")
>>> f.write("Does it works ?")
15
>>> f.close()
>>> os.listdir()
['test.txt']
```

When resetting I get the following message _The filesystem appears to be corrupted. [...]._ from inisetup.py

I tried to erase and reflash everything on two different boards with the same results

Wind-stormger, 2023-01-11:
@k-pa @0xabe1 @FuHu-QC 

 This PR #10462 will definitely fix this issue.

HonestQiao, 2023-01-26:
I have the same problem on ESP32-S2-Mini-1.

**chip_id:**
Chip is ESP32-S2FH32
Features: WiFi, Embedded 4MB Flash, 105C temp rating, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:97:f1:1c
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB

**TinyUF2 infomation:**
TinyUF2 Bootloader 0.12.0-2-ga4736a1-dirty - esp-idf (v4.4-dev-3608-gbbe2a1bf34) tinyusb (0.12.0-203-ga4cfd1c69)
Model: EETREE ESP32-S2
Board-ID: ESP32S2-PICO-V1.0
Date: Jan 26 2023

**Micropython:**
MicroPython v1.19.1-837-g67fac4ebc-dirty on 2023-01-26; EETREE_ESP32-S2 with ESP32-S2FH4
Type "help()" for more information.
>>> 
[image]

Wind-stormger, 2023-01-29:
@HonestQiao After repeated experiments, EPS32S3 does not have this problem. However, the ESP32S2 reproduces your problem, which is noteworthy.

HonestQiao, 2023-03-06:
> @HonestQiao After repeated experiments, EPS32S3 does not have this problem. However, the ESP32S2 reproduces your problem, which is noteworthy. There may be a better solution, and I will follow up.

Do you have any updates?

Wind-stormger, 2023-06-25:
https://github.com/micropython/micropython/pull/11864

closed 2023-06-25 by Wind-stormger