micropython/micropython #8458: @micropython.native StoreProhibited

stephanelsmith, 2022-03-28:
I recently updated my repo versions and my application started crashing.  I've tracked it down to the use @micropython.native.

1. I am using
- esp-idf v4.4
- The official esp ESP32 S3 dev board

2. The issue I'm seeing is a crash in @micropython.native functions.  
```
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core  1 register dump:
PC      : 0x420303bb  PS      : 0x00060730  A0      : 0x820f5952  A1      : 0x3fce40f0
A2      : 0x3fce4164  A3      : 0x4203cc85  A4      : 0x00000000  A5      : 0x3fce4200
A6      : 0x3fce4300  A7      : 0x3fce4320  A8      : 0x8200974d  A9      : 0x3fce4150
A10     : 0x3fca3030  A11     : 0x4203cc84  A12     : 0x3de11550  A13     : 0x00000002
A14     : 0x3c110ec8  A15     : 0x3c145110  SAR     : 0x0000001c  EXCCAUSE: 0x00000003
EXCVADDR: 0x4203cc84  LBEG    : 0x40056f5c  LEND    : 0x40056f72  e4b20 0x420221de:0x3fce4b40 0x4202252e:0x3fce4bd0 0x42022559:0x3fce4c00 0x4200707f:0x3fce4c20
ELF file SHA256: 456e5777abc30ba7
```

3. No issues as late as commit 64bfaae7ab33e628f28ca3b53b10893fb047b48e .  I have build errors between commits f2040bfc7ee033e48acef9f289790f3b4e6b74e5  and 0a217624e1d26899986d3e79eb604652da540b00 .  From commit 5c46721a1cc32bdb8f700955f9b441a08056f7b0  to most recent commit, I see issues when using @micropython.native.
....
ISSUE_OBSERVED most recent.
ISSUE_OBSERVED 5c46721a1cc32bdb8f700955f9b441a08056f7b0 tools/mpy-tool.py: Fix frozen comment generation to escape chars.
BUILD_FAIL      0a217624e1d26899986d3e79eb604652da540b00 tools/upip.py: Remove unused op_basename() function.
BUILD_FAIL      546e213265872cc6393af83a94bb2af5092d4348 esp32/boards: Add specific deploy instructions for S3 variant.
BUILD_FAIL      c2e8a5acd267e55a11a709d44f0385d0a40f3ec9 windows: Unify project file headers.
BUILD_FAIL      d6564a315967f80aafaa4c117892d0b21b0cb2bc tools/mpremote: Add "umount" command.
BUILD_FAIL      ad1f523e7ebae5f260e9c586b28e0d75d715cf65 tools/mpremote: Add "resume" and "soft-reset" commands.
BUILD_FAIL      414b59d39c9f37c0da5ee00d270aa630c392c2fd qemu-arm: Add tests for freezing viper and asm_thumb code.
BUILD_FAIL      0a2895b0998bc7ab66b8339acf64fda0339d1e53 tests/perf_bench: Skip bm_chaos test if random.randrange is unavailable.
BUILD_FAIL      66fc0f45c17c8f3af68d688a432bb51318bc434e tools/ci.sh: Run urandom test scripts as part of native module tests.
BUILD_FAIL      73a1927fce29f461945708675b82bd3d29f9f568 github/workflows: Add new workflow to test .mpy file format and tools.
BUILD_FAIL      c0f2af4e862a9bb80e401894bad30378dbeac85e ports: Recompile bytecode tests now that .mpy format changed.
BUILD_FAIL      f2040bfc7ee033e48acef9f289790f3b4e6b74e5 py: Rework bytecode and .mpy file format to be mostly static data.
LAST_WORKING    64bfaae7ab33e628f28ca3b53b10893fb047b48e esp32/README.md: Update list of supported IDF versions.

5. Work arounds are 
- Use commit 64bfaae7ab33e628f28ca3b53b10893fb047b48e or earlier.
- Comment out @micropython.native on commits after 64bfaae7ab33e628f28ca3b53b10893fb047b48e.

7. The issue seems to be environment specific, which means I don't have a simple test case...  A sample function which gave me an error is below.  **Direct test cases are not having issues**.  However when embedded in my larger application, I will have an error 100% of the time at the same location.  Adding seemingly meaningless lines of code (like print statements) and other nop lines will alleviate the issue.

```
@micropython.native
def filter(coll, itertee):
    return [el for el in coll if itertee(el)]  #error here

#test case which passes in a simple direct case on repl.  Fails when integrated in larger application.  No issue when using commit 64bfaae7ab33e628f28ca3b53b10893fb047b48e  or earlier or the decorator commented out.
filter(range(10),lambda x:x<5)
```

dpgeorge, 2022-03-29:
Thanks for the detailed bug report.  It's really good that you are able to test out unstable (ie non-release) firmware, because that helps to find very subtle bugs like this one!  If not for this report, the bug would have made its way into the next release.

Here is a minimal reproduction of the issue:
```python
# create 14 constants
print("0=================")
print("1=================")
print("2=================")
print("3=================")
print("4=================")
print("5=================")
print("6=================")
print("7=================")
print("8=================")
print("9=================")
print("10================")
print("11================")
print("12================")
print("13================")

# constant 14: mp_fun_table
# constant 15: prelude for filter
# constant 16: prelude for inner list comprehension, will map back to 0
@micropython.native
def filter(coll, itertee):
    print("start")
    return [el for el in coll if itertee(el)]

print(filter(range(10), lambda x: x < 5))
```

The bug is that the xtensa assembler (built into MicroPython) was using a narrow instruction to encode an `l32i` with an argument >=16, which is truncated via `arg & 0xf`, leading to bad code.

dpgeorge, 2022-03-29:
See #8462 for a fix.

stephanelsmith, 2022-03-29:
Wow!  I thought this one would be a doozy to reconcile since I couldn't provide a concrete test case.  Exciting.  Look forward to trying out the minimal reproduction (intriguing code snippet) and the update.

I am trying to stay as up to date as possible as I'm trying to work through S3 migration, both hardware and software.

I sincerely appreciate your eyes reviewing this.  It's clear I would never have been able to get to the bottom of it!  Thank you again.

closed 2022-03-29 by stephanelsmith