micropython/micropython #8460: Using cryptolib  on rp2 port

mrpackethead, 2022-03-29:
I want to be able to use a cryptolib.aes on board with rp2.    It doe'snt appear to be availalbe in the rp2 port, but it is in other ports.

How do i compile it into the rp2 port

dpgeorge, 2022-03-30:
This is not straightforward to do, because the rp2 port doesn't have an SSL library configured for it.

dpgeorge, 2022-03-31:
It will be possible to build in mbedtls or axtls, which are both included in this project, and that will allow building `ucryptolib`.  But to configure one of these libraries requires a bit of work.

dpgeorge, 2022-03-31:
See #8476.

mrpackethead, 2022-03-31:
Oh Wow, thats so kind of you!

mrpackethead, 2022-04-07:
>>> help("modules")
__main__          framebuf          uasyncio/funcs    ujson
_boot             gc                uasyncio/lock     umachine
_boot_fat         math              uasyncio/stream   uos
_onewire          micropython       ubinascii         urandom
_rp2              neopixel          ucollections      ure
_thread           onewire           ucryptolib        uselect
_uasyncio         rp2               uctypes           ustruct
builtins          uarray            uerrno            usys
cmath             uasyncio/__init__ uhashlib          utime
dht               uasyncio/core     uheapq            uzlib
ds18x20           uasyncio/event    uio
Plus any modules on the filesystem
>>> from ucryptolib import aes
>>> 

Test and works well thanks!

binarweb, 2022-06-14:
Between releases, there are a lot of changes. Using `v1.14` of micropython to port `ucryptolib` for rp2 might result in different issues (I haven't tested the port yet).
Is there a way that `ucryptolib` could be integrated in the current master branch and a new release be made available?

jimmo, 2022-06-15:
> Is there a way that ucryptolib could be integrated in the current master branch

@binarweb It is already in the master branch (see earlier comments). A release is coming soon, but in the meantime you can use the nightly builds.

binarweb, 2022-06-15:
Thank you!
Somehow I overlooked those firmware files.
I managed to encrypt and decrypt data successfully.

For anyone interested, the latest firmware for RPi Pcico is available [here](https://micropython.org/download/rp2-pico/).
For examples, I used [this](https://forum.micropython.org/viewtopic.php?t=6726).