micropython/micropython #8465: ESP32 related issue

GanerCodes, 2022-03-30:
I'm at a loss here, I can't seem to figure this out. I have this extremely odd issue where an LED controller I have will become laggy over the peirod of several days. This wouldn't be a problem if the issue didn't continue even after running machine.reset(). If I physically press the reset button on my esp32 (or unplug and replug) it fixes it. If anyone has any suggestions I'll be happy to hear them as I've been pondering this for months

peterhinch, 2022-03-30:
I suggest you post this query, with a minimal code sample which displays the problem, in [the forum](http://[forum](http://forum.micropython.org/index.php).micropython.org/index.php).

GanerCodes, 2022-03-30:
> Memory leak?

perhaps, but shouldn't machine.reset() reset the memory?

GanerCodes, 2022-03-30:
Is there any way to trigger a hardware reset? what all needs to be reset?

Could the line `(time.time_ns() / 1_000_000) % (2 ** 32)` have anything to do with it? Does it become inaccurate after time, even after machine reset?

jej, 2022-03-30:
Be sure to use `ticks_diff` in case (https://docs.micropython.org/en/latest/library/time.html#time.ticks_diff)

GanerCodes, 2022-03-30:
time wrap around doesn't have any effect on my code so I don't think that would do anything

GanerCodes, 2022-03-31:
im so stupid i needed integer division not normal division

implicit float casting is ruining my sanity

closed 2022-03-31 by GanerCodes