micropython/micropython #8473: kwargs error in recent py/runtime commits

stephanelsmith, 2022-03-31:
Updating to master, noticed a new eror relating to kwargs.   Last working commit is bb7087411.  On commit 1e99d29f3, I get a core dump, and 783b1a868 onwards, I get a TypeError.

ERROR            2e3f2045f (HEAD -> master, origin/master, origin/HEAD) py/runtime: Use size_t/ssize_t instead of uint/int.
47685180f tests/basics/fun_callstardblstar: Add coverage test.
9b74d71aa py/runtime: Drop new_alloc < 4 check.
3679a47eb py/runtime: Do not overallocate when len is known.
ERROR            783b1a868 py/runtime: Allow multiple *args in a function call.
LoadProhibited   1e99d29f3 py/runtime: Allow multiple **args in a function call.
PASS             bb7087411 py/vm: Prevent array bound warning when using -MP_OBJ_ITER_BUF_NSLOTS.

Here's the test case.  This schtik on this one is direct instantiation of kwargs is working fine. Which is why it probably passed test cases (and why this was a little tricky to find).  But when you pass kwargs through a function, I see the issue.

```
def kwargs_test1(kwargs):
    KwargsTest(**kwargs)
class KwargsTest():
    def __init__(self, hello            = None,
                       world            = False,
                       ):
        pass

kwargs = {'hello':True}

#direct instantiation works across commits
KwargsTest(**kwargs)
# <KwargsTest object at 3d832510>

#calling via function causes error on new commits
kwargs_test1(kwargs=kwargs)
# on commit bb7087411 and earlier, no error.
# on commit 1e99d29f3 core panic
# on commit 783b1a868 and owwards, I get this:
# Traceback (most recent call last):
  # File "<stdin>", line 1, in <module>
  # File "main.py", line 189, in kwargs_test1
# TypeError: can't convert dict to int

```

dlech, 2022-03-31:
Is the code compiled with `mpy-cross` first?

Do the proposed changes in https://github.com/micropython/micropython/pull/8472 fix the problem?

stephanelsmith, 2022-03-31:
On master branch, ran at root
make -C mpy-cross

Rebuilding, now I a build error
```
Traceback (most recent call last):
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 1609, in <module>
    main()
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 1584, in main
    compiled_modules = [read_mpy(file) for file in args.files]
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 1584, in <listcomp>
    compiled_modules = [read_mpy(file) for file in args.files]
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 1262, in read_mpy
    raw_code = read_raw_code(reader, cm_escaped_name, qstr_table, obj_table, segments)
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 1215, in read_raw_code
    read_raw_code(reader, cm_escaped_name, qstr_table, obj_table, segments)
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 1191, in read_raw_code
    rc = RawCodeNative(
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 913, in __init__
    super(RawCodeNative, self).__init__(
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 735, in __init__
    ) = extract_prelude(self.fun_data, prelude_offset)
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 423, in extract_prelude
    ) = read_prelude_sig(local_read_byte)
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 371, in read_prelude_sig
    z = read_byte()
  File "/home/ssmith/micropython/tools/mpy-tool.py", line 411, in local_read_byte
    b = bytecode[ip_ref[0]]
IndexError: index out of range
```

Will look at proposed change here shortly.

dlech, 2022-03-31:
> ran at root

Why as root?

> Was error reproducable on your end?

No, I copied and pasted the example from the first comment and ran it in a unix 64-bit build without error.

stephanelsmith, 2022-03-31:
Not as root, root micropython directory.

dlech, 2022-03-31:
The mpy-tool.py error could be due to other recent changes like #8191 or #8433.

stephanelsmith, 2022-03-31:
Yeah, I recently worked on a byte code error, https://github.com/micropython/micropython/issues/8458
Trying to get revert back to last working mpy-cross now...

Could very well be related if there was additional byte code format change between https://github.com/micropython/micropython/commit/35dbde163a8a253679c09b4646a9177cde5345f2  and the latest commit.

Will keep looking for esp32.

stephanelsmith, 2022-03-31:
I'm able to re- "make -C mpy-cross" from this commit
https://github.com/micropython/micropython/commit/35dbde163a8a253679c09b4646a9177cde5345f2 
Is fairly new, only 4 days old.  Why it's not working on latest sounds like a confounding/contriubting factor. 

For esp32 work-around, https://github.com/micropython/micropython/commit/bb70874111dbb246624a68c013e8f1c3245ca0d8 is good. Alternatively, don't pass down kwargs on newer commits.

dpgeorge, 2022-04-01:
Please can you re-try with latest master, now that #8472 is merged.  If it still fails, please give exact steps to reproduce (and try without any `@micropython.native`).

stephanelsmith, 2022-04-01:
Confirming that with all  `@micropython.native`  commented out
- Build error above goes away
- The weird kwargs issue also goes away

Looks like `@micropython.native` is it!
 
Additional info:
- Pulled master (8baf05af8 (HEAD -> master, origin/master, origin/HEAD) py/makeqstrdefs: Cleanup and extend source file classification.)
- Re-made mpy-cross, clean build.
- Running on custom S3 board

Finally, (getting ahead of myself),  `@micropython.native` I've never got working on async functions.  I don't think it's an "issue" but just wanted to put it on the radar!

I'll keep working on master and comment out all `@micropython.native` which is fine for me.  I'll close both issues on my end here too.

stephanelsmith, 2022-04-01:
Also, @dlech and @dpgeorge huge thank you .  This is tricky business.   I sincerely appreciate your taking up my questions and your patience as I work through on my end.

closed 2022-04-02 by stephanelsmith